BeamNG/BeamNGpy #242: Cannot find weather presets (in art/weather/defaults.json)

jakobfrosch, 2023-12-29:
According to the documentation, the example json objects defining weather presets can be found in BeamNG.tech’s art/weather/defaults.json file. Also with searching trough all my files, i cannot find that defaults.json file. Can someone please assist here?
I´m currently still using BeamNG.tech 0.30 | BeamNG.tech 1.27.1. But I can also not find the file in the newly released version.

aivora-beamng, 2024-01-02:
Hi, the file is located under `BeamNG.tech.<version>\gameengine.zip\art\weather\defaults.json`. The available values are `sunny_evening`, `sunny`, `cloudy_evening`, `sunny_noon`, `rainy`, `foggy_night`, `foggy_morning`.

You can also get the list of all available weather presets by running `core_weather.getPresets()` in the simulator console:
[image]

jakobfrosch, 2024-01-02:
Thank you! That helpes a lot. 
Ticket closed!

closed 2024-01-02 by jakobfrosch