BeamNG/BeamNGpy #244: Starting Freeroam

uz1me, 2024-01-14:
I'm intending to explore off-road environments in BeamNG.tech. So I download some mods and intend to run in them. But the mods can not be detected by `` seems only avalible in Freeroam. How should I start Freeroam with Python scripts?

aivora-beamng, 2024-01-17:
Hi,
you can start freeroam by running following code:
```python
beamng.queue_lua_command("freeroam_freeroam.startFreeroamByName('italy')") # start freeroam on the Italy map
```

We may add an API for that if it's useful, however, I'm not sure if freeroam is needed in your case. What do you mean by "mods can not be detected by ``"?. What is the issue here, do you have code to reproduce? Mods should be compatible with BeamNGpy, what are the mods you are installing?

uz1me, 2024-01-17:
您好，您的来信我已收到，将在第一时间向您回复。谢谢。        ——杨溢

uz1me, 2024-01-18:
My apologies, I was intending to say the mods can not be detected by `beamng.get_scenarios()`.

uz1me, 2024-01-18:
My apologies, I was intending to say the mods can not be detected by `beamng.get_scenarios().`
 I'm installing mods of terrains and custom maps, and I want to explore the map freely. So i thought i might need to start freeroam. 
But later, i unpacked the mod and copy contents in `levels` subfolder into the beamng path like `./BeamNG.tech/levels`. Then I use try to load a undefined scenario with codes like `Scenario(level='mod_name', name='test')`. It works and creates a scenario named 'test'. So I guess my problem is solved. The `name` in `Scenario` does not has to be a predefined scenario, it could be created under any levels that can be detected with `beamng.get_levels()`.

Many thanks!

closed 2024-01-18 by uz1me