BeamNG/BeamNGpy #247: How to Simulate Drunk Driving

Guaziboi, 2024-01-31:
So I'm working with my university to develop a driving simulator that has the ability to have a "drunk driving" mode.  Currently using OpenTrack to track the users head movements and this works perfectly. I feel like it would be easier to modify BeamNG to have the desired effect. Not sure what the best course of action is to accomplish my task. Any help is appreciated!

aivora-beamng, 2024-02-05:
Hi,
that depends on the definition of what the "drunk driving" mode should include. To have a shakey camera for example, you could start with the `shake` camera mode implemented in `lua\ge\extensions\core\cameraModes\shake.lua`. What are the features that your "drunk driving" mode is expected to include?

Guaziboi, 2024-02-05:
It just needs the general wavy/heavy head movement would be fine and it would need to be toggled on and off.

How do I activate the shake camera mode? Doesn't look like it's available within the game settings but I did find it inside the file.

aivora-beamng, 2024-02-06:
Hi, it's a camera filter, so you need to use it with another mode. I guess you're using the driver camera, so I edited it to use the `shake` filter and I attach it here:
[driver_shake.zip](https://github.com/BeamNG/BeamNGpy/files/14177565/driver_shake.zip)

You can unpack the folder to `%localappdata/BeamNG.{drive/tech}/0.31/mods/unpacked` and it should be recognized as a mod by BeamNG. You can then edit the parameters of the shake filter in the .lua file directly. To toggle it off and on, the easiest solution would be to switch with the normal camera (you can disable all the other modes in the settings).

Guaziboi, 2024-02-20:
So as soon as I unpack the folder in the correct spot, would I see a difference in the camera behavior right away or would I need to adjust the values first? I've already unpacked it and there doesn't seem to be a difference.

aivora-beamng, 2024-02-21:
Hi, if you unpacked it correctly, then you should see the `Driver_shake` camera as one of the camera modes, and you can switch to it using the standard keyboard shortcut (`C`):

[image]
[image]

AbdelrahmanElsaidElsawy, 2024-08-29:
Hi,

As there hasn’t been any further activity on this issue, I’m going to close the ticket for now. If you encounter any more problems or have additional questions, feel free to reopen the issue or start a new one. We’re here to help!

Thank you for using BeamNGpy.

Best regards,

closed 2024-08-29 by AbdelrahmanElsaidElsawy