BeamNG/BeamNGpy #250: ConnetionResetError when retrieving sensor state via

Leviathan321, 2024-02-22:
I am using BeamNG 0.23.5 with beamngpy==1.22. I have spawned a vehicle and wanted to retrieve the sensor state. When I call **vehicle.poll_sensors()** I get the following error:

`sensors = self.vehicle.poll_sensors()
  File "XXX\venv\lib\site-packages\beamngpy\vehicle.py", line 360, in poll_sensors
    response = self.bng.recv()
  File "XXX\venv\lib\site-packages\beamngpy\beamng.py", line 557, in recv
    return recv_msg(self.skt)
  File "XXX\venv\lib\site-packages\beamngpy\beamngcommon.py", line 256, in recv_msg
    length = skt.recv(16)
ConnectionResetError: [WinError 10054] Eine vorhandene Verbindung wurde vom Remotehost geschlossen
[Vehicle] Disconnecting.
`
I checked the firewall already, and I checked that there is a port allocated for the vehicle. Does anyone have any ideas?

aivora-beamng, 2024-02-23:
Hi, the supported version for BeamNG 0.23 is `beamngpy==1.21.1`, can you try with this version?

Leviathan321, 2024-02-24:
Hi, yes I have tried both versions and I get the same problem. I checked this script [here](https://github.com/BeamNG/BeamNGpy/blob/v1.22/examples/multishot_camera.ipynb), and that works for me. However, when I add a userpath to the initialization of BeamNGpy or change the scenario to my custom scenario, I get the same error.

I dont understand how this can lead to this error.

aivora-beamng, 2024-02-26:
Do you see any errors in BeamNG console? Can you please share the `beamng.log` files that are generated by BeamNG in your userpath? Also, is your userpath "clean enough"? There may for example be some mod files which are causing BeamNG to not work correctly.

Leviathan321, 2024-02-28:
The problem is now gone, after I have cleaned the user folder and switched also to the 1.21.1 version. I am not sure if this was the reason, when I find time, I will check that in detail.

closed 2024-02-28 by Leviathan321