BeamNG/BeamNGpy #252: Example needed: taking a bird-view picture of the map from an absolute location

alessiogambi, 2024-03-15:
Currently, one can attach cameras to a vehicle and take a birdview picture of the map, but setting the relative position of the camera to something (0, 0, 30) and the forward direction of the camera to (0, 0, -1). So, the camera will be 30 meters above the car pointing down. Doing so that the camera will rotate along with the vehicle. How could one simulate something like a freeCamera? Basically, a camera that does not move and does not rotate along the vehicle it is attached to?

For the position, one can probably compute the distance between the car and the (expected) position of the camera, but for the rotation I am kinda lost.

A code example would help clarify the reasoning behind this.

> PS: I looked at the multishot camera, but that does not help since the vehicle never moves

aivora-beamng, 2024-03-15:
Hi, you can use the `is_static` property of the Camera sensor to achieve that:
```python
camera = Camera('camera1', beamng, vehicle, is_static=True, pos=(0, 0, 30), dir=(0, 0, -1))
```
[image]

alessiogambi, 2024-03-15:
Super! Thanks @aivora-beamng

closed 2024-03-18 by aivora-beamng