BeamNG/BeamNGpy #255: Unable to find a way to get the car in free roam

emmettprexus, 2024-03-22:
Hello!

I have managed to get the player vehicle if I am in a scenario that was either loaded from the game or created programmatically:

```
            vehicle_id = beamng.vehicles.get_player_vehicle_id().get("vid")
            vehicle = beamng.scenario.get_vehicle(vehicle_id)

            vehicle.sensors.attach("electrics", Electrics())
            vehicle.sensors.poll()
```

This does not work in free roam mode though, as it complains that there is no scenario there... How would I get the current vehicle in free roam? I cannot spawn the vehicle from my program, it needs to simply listen on what's currently in the game and use that vehicle.

aivora-beamng, 2024-03-22:
Hi,
you should be able to use the method `beamng.vehicles.get_current()` (https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.api.beamng.VehiclesApi.get_current). Let me know if it works for your usecase!

emmettprexus, 2024-03-22:
Excellent! I had tried the get_current() before, but couldn't get it to work - the connect() was the solution! Thanks a lot! I have another issue regarding the correct flow for handling change of cars or map - but I'll post that to a new issue not to confuse this one.

closed 2024-03-22 by emmettprexus