BeamNG/BeamNGpy #256: Detecting and reconnecting to new player vehicle

emmettprexus, 2024-03-22:
My script fails when I'm changing vehicle in game. After restarting the script, everything works fine, so there are something I must be doing wrong - not sure if I should disconnect the vehicle, beamng or what the trick here is:

```from beamngpy import BeamNGpy, Scenario, Vehicle
from beamngpy.sensors import Electrics

import time

beamng = None
vehicle = None
connected = False

while True:
    if not connected:
        print("Attempting to connect to BeamNG")

        try:
            beamng = BeamNGpy('localhost', 64256, home='C:\\Program Files (x86)\\Steam\\steamapps\\common\\BeamNG.drive')
            beamng.open()

            while vehicle is None:
                vehicle = beamng.vehicles.get_current().get('thePlayer')

                if vehicle is None:
                    print("No vehicle found, trying again soon")
                    time.sleep(5)

            vehicle.connect(beamng)

            vehicle.sensors.attach("electrics", Electrics())

            connected = True

            print("Connection successful")

        except ConnectionRefusedError:
            print("Failed to connect")

    if connected:
        try:
            vehicle.sensors.poll()

            lights_on = vehicle.sensors['electrics']['lights'] > 0

            print("Lights on: " + str(lights_on))

        except ConnectionRefusedError:
            print("Lost connection to beamng")
            connected = False

    time.sleep(1)
```

The error looks like this:

```
Lost connection to beamng
Attempting to connect to BeamNG
Error connecting to BeamNG.tech vehicle thePlayer. 25 tries left.
[WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):
  File "C:\Python\Python312\Lib\site-packages\beamngpy\connection\connection.py", line 106, in connect_to_vehicle
    self.skt = PrefixedLengthSocket(self.host, self.port)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python\Python312\Lib\site-packages\beamngpy\connection\prefixed_length_socket.py", line 52, in __init__
    self.skt.connect((host, port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):
  File "C:\Python\Python312\Lib\site-packages\beamngpy\connection\connection.py", line 106, in connect_to_vehicle
    self.skt = PrefixedLengthSocket(self.host, self.port)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python\Python312\Lib\site-packages\beamngpy\connection\prefixed_length_socket.py", line 52, in __init__
    self.skt.connect((host, port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\emmet\Repos\BeamNG Light Controller\orchestrator\example.py", line 25, in <module>
    vehicle.connect(beamng)
  File "C:\Python\Python312\Lib\site-packages\beamngpy\vehicle\vehicle.py", line 221, in connect
    self.connection.connect_to_vehicle(self)
  File "C:\Python\Python312\Lib\site-packages\beamngpy\connection\connection.py", line 112, in connect_to_vehicle
    sleep(5)
```

Can anyone assist in fixing this logic? Basically I just need to constantly monitor the current vehicle in the game - or wait for one to spawn. That's it.

emmettprexus, 2024-03-25:
Turns out there was a little flaw in the logic. When I'm doing:

```
        except ConnectionRefusedError:
            print("Lost connection to beamng")
            connected = False
```

...I should remember to clear the vehicle variable as well like this:

```
        except ConnectionRefusedError:
            print("Lost connection to beamng")
            connected = False
            vehicle = None
```

Then it'll correctly grab the new vehicle. I also managed to change it so it doesn't reconnect to beamng by attempting to reconnect to the new 'current vehicle' and if that fails it'll try and reconnect to beamng.

Remaining question is if there is a way to lower the timeout before the `.poll()` throws because the vehicle is no longer present. It seems to be a 10 second timeout, and it would be nice if it was a bit quicker! :)

robertpimlott, 2024-04-24:
hi im slighting having a issue trying to detect any vehicle

import beamngpy

def detect_player_vehicle(scenario):
    # Get the list of vehicles in the scenario
    vehicles = scenario.vehicles

    # Iterate through the vehicles to find the player's vehicle
    for vehicle_name, vehicle in vehicles.items():
        if vehicle.control == 'player':
            return vehicle_name

    return None

if __name__ == "__main__":
    # Define host and port
    host = "******"
    port = ******

    # Connect to BeamNG.tech
    beamng = beamngpy.BeamNGpy(host, port, home='C:\\BeamNG.tech.v0.31.3.0')

    # Launch BeamNG.tech
    beamng.open()

    # Wait for BeamNG.tech to load
    input("Press Enter after BeamNG.tech is fully loaded and the Lua console is ready...")

    # Get the current scenario
    current_scenario = beamng.get_current_scenario()

    if current_scenario is None:
        print("No scenario loaded.")
    else:
        print("Current scenario:", current_scenario)

        # Detect the player's vehicle
        vehicle_name = detect_player_vehicle(current_scenario)
        if vehicle_name:
            print("Player vehicle detected:", vehicle_name)
        else:
            print("Player vehicle not found.")

    # Close BeamNG.tech connection
    beamng.close()

    # Reconnect to BeamNG.tech
    beamng.open()

    # Wait for BeamNG.tech to load again
    input("Press Enter after BeamNG.tech is fully loaded and the Lua console is ready...")

    # Continue with further actions if needed 
    
    
    ____________________
    
    
    as the print says all im getting is (Player vehicle not found)

closed 2024-04-25 by emmettprexus