BeamNG/BeamNGpy #257: why my vehicle don't move?

lsm2842035890, 2024-03-25:
[image]
[image]
I loaded the gridsmall map and paved a simple road. I want the vehicle to run according to my trajectory. Why does the vehicle not move?

ASionville, 2024-03-25:
For now I think your "AI" stops with the python script.
You should add `time.sleep(x)` at the end of script for it to be "alive" for x more seconds and actually move.
You can put it in a while loop for better control.
You can consider using `bng.step(n)` too if you want, n here is the number of steps to go ahead to if I remember correctly

lsm2842035890, 2024-03-25:
044.235|E|libbeamng.Lua.Exception|*** FATAL LUA ERROR:
044.236|E|libbeamng.Lua.Exception|lua/vehicle/ai.lua:4537: cannot convert 'nil' to 'double'
044.236|E|libbeamng.Lua.Exception|=============== Stack Traceback >> START >>
044.237|E|libbeamng.Lua.Exception|(1)  C function 'function: 0x02bc6a4185a8'
044.237|E|libbeamng.Lua.Exception|(2) global C function 'vec3'
044.238|E|libbeamng.Lua.Exception|(3) Lua field 'driveUsingPath' at file 'lua/vehicle/ai.lua:4537'
044.241|E|libbeamng.Lua.Exception|  Local variables:
044.242|E|libbeamng.Lua.Exception|  arg = table: 0x02bc5909c470  {wpSpeeds:table: 0x02bc5909c190, script:table: 0x02bc5909c148}
044.243|E|libbeamng.Lua.Exception|  script = table: 0x02bc5909c148  {1:table: 0x02bc5909c208, 2:table: 0x02bc5909c280, 3:table: 0x02bc5909c360 (more...)}
044.243|E|libbeamng.Lua.Exception|  dir = nil
044.244|E|libbeamng.Lua.Exception|  up = nil
044.245|E|libbeamng.Lua.Exception|  pos = nil
044.245|E|libbeamng.Lua.Exception|(4) Lua local 'handler' at file 'lua/vehicle/extensions/tech/techCore.lua:278'
044.246|E|libbeamng.Lua.Exception|  Local variables:
044.247|E|libbeamng.Lua.Exception|  request = table: 0x02bc6a350060  {_id:1, handled:false, type:SetAiLine, skt:tcp{client}: 000002BC29497670 (more...)}
044.247|E|libbeamng.Lua.Exception|  nodes = table: 0x02bc6a3501f0  {1:table: 0x02bc6a350270, 2:table: 0x02bc6a345890, 3:table: 0x02bc6a345af8 (more...)}
044.248|E|libbeamng.Lua.Exception|  fauxPath = table: 0x02bc5909c148  {1:table: 0x02bc5909c208, 2:table: 0x02bc5909c280, 3:table: 0x02bc5909c360 (more...)}
044.248|E|libbeamng.Lua.Exception|  cling = boolean: true
044.249|E|libbeamng.Lua.Exception|  z = number: 0
044.249|E|libbeamng.Lua.Exception|  speedList = table: 0x02bc5909c190  {1:30, 2:30, 3:30}
044.250|E|libbeamng.Lua.Exception|  arg = table: 0x02bc5909c470  {wpSpeeds:table: 0x02bc5909c190, script:table: 0x02bc5909c148}
044.251|E|libbeamng.Lua.Exception|(5) Lua upvalue 'handleRequest' at file 'lua/common/tech/techCommunication.lua:129'
044.251|E|libbeamng.Lua.Exception|  Local variables:
044.252|E|libbeamng.Lua.Exception|  handler = Lua function '(anonymous)' (defined at line 251 of chunk @lua/vehicle/extensions/tech/techCore.lua)
044.253|E|libbeamng.Lua.Exception|  request = table: 0x02bc6a350060  {_id:1, handled:false, type:SetAiLine, skt:tcp{client}: 000002BC29497670 (more...)}
044.253|E|libbeamng.Lua.Exception|(6) Lua field 'checkMessages' at file 'lua/common/tech/techCommunication.lua:162'
044.254|E|libbeamng.Lua.Exception|  Local variables:
044.258|E|libbeamng.Lua.Exception|  E = table: 0x02bc6a43c308  {__extensionPath__:tech/techCore, handleExecuteScript:function: 0x02bc58ac5a58 (more...)}
044.258|E|libbeamng.Lua.Exception|  clients = table: 0x02bc103fe520  {1:tcp{client}: 000002BC29497670}
044.259|E|libbeamng.Lua.Exception|  message = string: "劋type㏒etAiLineine搨os揬0蚛1怽0peed\30偅pos揬0蚛1鬨0peed\30偅pos揬0蚛2X\0peed\30ling茫_id\1"
044.260|E|libbeamng.Lua.Exception|  readable = table: 0x02bc6a34fdc8  {1:tcp{client}: 000002BC29497670, tcp{client}: 000002BC29497670:1}
044.260|E|libbeamng.Lua.Exception|  writable = table: 0x02bc6a34fe48  {1:tcp{client}: 000002BC29497670, tcp{client}: 000002BC29497670:1}
044.261|E|libbeamng.Lua.Exception|  err = nil
044.261|E|libbeamng.Lua.Exception|  ret = boolean: true
044.262|E|libbeamng.Lua.Exception|  (for index) = number: 1
044.263|E|libbeamng.Lua.Exception|  (for limit) = number: 1
044.263|E|libbeamng.Lua.Exception|  (for step) = number: 1
044.264|E|libbeamng.Lua.Exception|  i = number: 1
044.264|E|libbeamng.Lua.Exception|  skt = tcp{client}: 000002BC29497670
044.265|E|libbeamng.Lua.Exception|  request = table: 0x02bc6a350060  {_id:1, handled:false, type:SetAiLine, skt:tcp{client}: 000002BC29497670 (more...)}
044.265|E|libbeamng.Lua.Exception|  msgType = string: "handleSetAiLine"
044.266|E|libbeamng.Lua.Exception|  handler = Lua function '(anonymous)' (defined at line 251 of chunk @lua/vehicle/extensions/tech/techCore.lua)
044.267|E|libbeamng.Lua.Exception|(7) Lua local 'func' at file 'lua/vehicle/extensions/tech/techCore.lua:187'
044.267|E|libbeamng.Lua.Exception|(8) Lua field 'hook' at file 'lua/common/extensions.lua:695'
044.268|E|libbeamng.Lua.Exception|  Local variables:
044.268|E|libbeamng.Lua.Exception|  funcName = string: "onDebugDraw"
044.269|E|libbeamng.Lua.Exception|  funcList = table: 0x02bc5b343ed0  {1:function: 0x02bc5b34e778}
044.270|E|libbeamng.Lua.Exception|  (for generator) = C function: builtin#6
044.275|E|libbeamng.Lua.Exception|  (for state) = table: 0x02bc5b343ed0  {1:function: 0x02bc5b34e778}
044.275|E|libbeamng.Lua.Exception|  (for control) = number: 1
044.276|E|libbeamng.Lua.Exception|  _ = number: 1
044.277|E|libbeamng.Lua.Exception|  func = Lua function '(anonymous)' (defined at line 170 of chunk @lua/vehicle/extensions/tech/techCore.lua)
044.277|E|libbeamng.Lua.Exception|(9) Lua function 'onDebugDraw' at file 'lua/vehicle/main.lua:129' (best guess)
044.278|E|libbeamng.Lua.Exception|  Local variables:
044.278|E|libbeamng.Lua.Exception|  x = number: -0.331062
044.279|E|libbeamng.Lua.Exception|  y = number: 301.495
044.279|E|libbeamng.Lua.Exception|  z = number: 1.19209e-07
044.280|E|libbeamng.Lua.Exception|--------------- << END <<
044.281|E|libbeamng.Lua.Exception|******************************************
044.281|E|libbeamng.Lua.Exception|*** DISABLING VEHICLE DUE TO EXCEPTION ***
044.282|E|libbeamng.Lua.Exception|******************************************
044.282|E|libbeamng.Lua.Exception|*** vehicle/object error
044.476|E|CEF.MainGEUI#local://local/ui/ui-vue/dist/index.js:1|TypeError: Cannot read property 'emitsOptions' of null
044.526|E|CEF.MainGEUI#local://local/ui/ui-vue/dist/index.js:1|TypeError: Cannot read property 'emitsOptions' of null
044.579|E|CEF.MainGEUI#local://local/ui/ui-vue/dist/index.js:1|TypeError: Cannot read property 'emitsOptions' of null

lsm2842035890, 2024-03-25:
I tried that, but I still can't get the vehicle to move according to my coordinates. Could it be related to the map? The map I loaded is gridsmall, and the road sign information in it is incomplete, making it impossible to move the vehicle.

ASionville, 2024-03-25:
If you put the AI in `span` mode, does it move at all ?

aivora-beamng, 2024-04-08:
Hi @lsm2842035890,
the script must be "realistic" (the vehicle AI tries to get the vehicle to the point `(x, y, z)` in the time `t`. You may use the [ScriptAI Editor](https://documentation.beamng.com/beamng_tech/scriptaieditor/) to create valid scripts. Also, the `looped` argument of the `road_a` road should be set to `False`, as your road is a straight one.

This code works for me, you can try yourself:
```python
from beamngpy import BeamNGpy, Road, Scenario, Vehicle

def generate_scripts():
    script = [
        {'x': 0, 'y': 300, 'z': 0.1, 't': 0},
        {'x': 0, 'y': 400, 'z': 0.1, 't': 10},
        {'x': 0, 'y': 500, 'z': 0.1, 't': 20},
        {'x': 0, 'y': 600, 'z': 0.1, 't': 30}
    ]
    return script

def set_scripts():
    bng = BeamNGpy('localhost', 64256)
    bng.open()

    scenario = Scenario('smallgrid', 'custom_road_network')

    road_a = Road('track_editor_C_center', rid='center_road', looped=False)
    road_a.add_nodes(
        (0, 300, 0, 6),
        (0, 400, 0, 6),
        (0, 500, 0, 6),
        (0, 600, 0, 6)
    )
    scenario.add_road(road_a)

    vehicle = Vehicle('ego_vehicle', model='etk800')
    scenario.add_vehicle(vehicle, pos=(0, 300, 0), rot_quat=(0, 0, 1, 0))
    positions = [(0, 400, 0), (0, 500, 0), (0, 600, 0)]
    scales = [(1., 1., 1.)] * 3
    ids = ['my_waypoint_1', 'my_waypoint_2', 'my_waypoint_3']
    scenario.add_checkpoints(positions, scales, ids)
    scripts = generate_scripts()
    scenario.make(bng)
    bng.scenario.load(scenario)
    bng.scenario.start()

    vehicle.ai.set_speed(30, 'set')
    vehicle.ai.execute_script(scripts)

set_scripts()
input('Press Enter to exit...')
```

lucena564, 2024-04-25:
I was with the same point. That function here:
start_recording()
stop_recording()
https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.api.vehicle.VehicleApi

It should record that points for us? Have some feacture that can helps with it? Or it is like try and error?

In ai_line.py we have points passed by sin function, like:
[image]

But if I load that map:
```
scenario = Scenario('west_coast_usa', 'Loop 1 da Rodovia')
```

It will load this:

[image]

I dont have the points to follow, but have someting to get the point at the moment to build it manualy? 

Was thinking about run that using:
```
vehicle.ai.set_mode('disabled') 
```

lucena564, 2024-04-26:
I got those points here:
```
script = [{'x': 460.86572265625, 'y': 328.2005615234375, 'z': 119.28526306152344, 't': 10}, {'x': 430.08746337890625, 'y': 301.0779724121094, 'z': 119.33838653564453, 't': 20.0}, {'x': 362.0964050292969, 'y': 250.7567138671875, 'z': 119.33403015136719, 't': 30}]
points = [(460.86572265625, 328.2005615234375, 119.28526306152344), (430.08746337890625, 301.0779724121094, 119.33838653564453), (362.0964050292969, 250.7567138671875, 119.33403015136719)]
sphere_coordinates = [(460.86572265625, 328.2005615234375, 119.28526306152344), (430.08746337890625, 301.0779724121094, 119.33838653564453), (362.0964050292969, 250.7567138671875, 119.33403015136719)]
sphere_radii = [0.0, 0.246201938253052, 0.08550503583141716]
sphere_colors = [(0.0, 0, 0, 0.8), (0.984807753012208, 0, 0, 0.8), (-0.34202014332566866, 0, 0, 0.8)]
```
```
scenario = Scenario('west_coast_usa', 'Loop 1 da Rodovia', description='Spanning the map with an advanced IMU sensor')
vehicle = Vehicle('ego_vehicle', model='etk800', license='RED', color='Red')
scenario.add_vehicle(vehicle, pos=(460, 328, 119.28), rot_quat=(0, 0, 0.3826834, 0.9238795))
```

But looks that AI is not looking collision, just going on to the left wall. Those points have time enough to go across.

[image]

[image]

lucena564, 2024-04-26:
I dont know if it will help anyone but..

The problem was that I was using the function to get my positions of the car on the street:
```
vehicle.get_bbox()
```
It returns:
```
{'front_bottom_left': [460.85791015625, 328.1919860839844, 119.28421783447266], 'front_bottom_right': [462.30120849609375, 326.7507019042969, 119.31969451904297], 'front_top_left': [460.831298828125, 328.1995544433594, 120.67394256591797], 'front_top_right': [462.27459716796875, 326.7582702636719, 120.70941925048828], 'rear_bottom_left': [464.15704345703125, 331.4969177246094, 119.32947540283203], 'rear_bottom_right': [465.600341796875, 330.0556335449219, 119.36495208740234], 'rear_top_left': [464.13043212890625, 331.5044860839844, 120.71920013427734], 'rear_top_right': [465.57373046875, 330.0632019042969, 120.75467681884766]}
```

The point is that the gridof the ground is a little bit hier then the grid of the car, so, when I passed the ```cling=True``` that makes the car try follow a really deep point (the first ground below the streat grid). Looks that:

[image]

It is possible observ that have a line that going deep in to the ground, it is the second point that the car should go to, was the point of the bug.

To fix that needed use an offset of ```0.9``` at my node in 'z' position.

```
node = {
            'x': (points['front_bottom_left'][0] + points['front_bottom_right'][0])/2,
            'y': (points['front_bottom_left'][1] + points['front_bottom_right'][1])/2,
            'z': points['front_bottom_left'][2], # + 0.9,
            't': (current- initialization).total_seconds()
        }
```

Using that we can get someting like that:

[image]

closed 2024-06-11 by aivora-beamng