BeamNG/BeamNGpy #258: BeamNG crashes with unavailable rendering device message.

Leviathan321, 2024-03-29:
I am using the research version of BeamNG with a custom scenario, and it crashes from time to time with the message:
**"It is possible that your hardware/OS does not meet the minimum system requirements."**

Also a browser tab tops up telling me that the directx version is not supported (I am using v11.0).

I have the following system properties:

GPU - NVIDIA 1080 TI
CPU - Intel i7 8700 3.2 Ghz
RAM - 32 GB

Any ideas?

aivora-beamng, 2024-04-02:
Hi, is your graphic driver updated to the latest version? Can you send the `beamng.log` file, which is located in the userfolder (`%localappdata%/BeamNG.{drive/tech}/<version>`)?

Leviathan321, 2024-04-02:
Yes I am using the latest driving for my graphics card. The log files created are attached...
[beamng.2.log](https://github.com/BeamNG/BeamNGpy/files/14840659/beamng.2.log)
[beamng.log](https://github.com/BeamNG/BeamNGpy/files/14840664/beamng.log)
[beamng-dxDiag.txt](https://github.com/BeamNG/BeamNGpy/files/14840666/beamng-dxDiag.txt)

Leviathan321, 2024-04-06:
Is there anything I could try? Do you have any ideas? When I am checking the resource usage, I see that the GPU is used.

aivora-beamng, 2024-04-08:
Hi, do you have a screenshot of the error? It does not seem to be an error produced by BeamNG, but by your operating system.

There is no information about the source of the issue in the log files attached, unfortunately. The problem might be in your operating system switching to the Intel UHD Graphics adapter in some cases. You could try setting BeamNG to always run on your NVIDIA GPU (https://www.intel.com/content/www/us/en/support/articles/000090168/graphics.html).

Leviathan321, 2024-05-02:
So I have verified that my GPU is used during the execution (using the performance monitor). Also in the log I see that my GPU is detected:

```
2024-05-02 18:03:02.769887: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1402] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 21509 MB memory) -> physical GPU (device: 0, name: NVIDIA GeForce RTX 4090, pci bus id: 0000:01:00.0, compute capability: 8.9)
2024-05-02 18:03:02.771951: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x24d29a75350 initialized for platform CUDA (this does not guarantee that XLA will be used). Devices:
2024-05-02 18:03:02.772126: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): NVIDIA GeForce RTX 4090, Compute Capability 8.9
...
Copying from [C:\Users\Lev\Documents\testing\testingautomated-usi\maxitwo\levels_template\tig] to [C:\Users\Lev\Documents\BeamNG_User\0.23\levels\tig]
```

Leviathan321, 2024-05-08:
This is the message I sometimes I get:
[image]