BeamNG/BeamNGpy #261: ScenarioObject name cannot be Empty or None

HLCaptain, 2024-04-08:
When trying to get objects from a Scenario, I always get the exception: `beamngpy.logging.BNGValueError: Name cannot be empty or None!`. On the map I loaded (east_coast_usa), some static objects does not have a name and `validate_object_name` throws an error while creating the `ScenarioObject` in python. I suspect, as the built-in world editor allows objects to have an Empty or None name, the python API should as well.

aivora-beamng, 2024-04-08:
Hi,
can you please post the code that causes the error? The `validate_object_name` function is only called if the name is not None, so I'm not sure what's happening here.

HLCaptain, 2024-04-12:
Hi, problematic source code available [here](https://github.com/HLCaptain/hawk-ai/blob/410d9e391b1a4d6709869ae0b316d4c49de981fb/data-generation/beamng-scripts/image_in_x_data_generation_strategy.py#L109C1-L109C17). You can start testing by running [this file](https://github.com/HLCaptain/hawk-ai/blob/main/data-generation/beamng-scripts/beamng_data_generation.py), then you should see some logs like this, after the level is loaded:
```
2024-04-12 21:06:46,987 |INFO     |beamngpy.BeamNGpy             |Successfully connected to BeamNG.tech.
2024-04-12 21:06:46,987 |INFO     |beamngpy.BeamNGpy             |Successfully connected to vehicle vehicle_38762.
2024-04-12 21:06:47,044 |INFO     |beamngpy.Camera               |Opened Camera: "vehicle_38762_camera"
Monitoring data for iteration 0
Snapping camera
Loading static instances from scenario...
Could not load static instances from scenario: Name cannot be empty or None!
```

Keep in mind, that I use `BNG_HOME` environment variable to find the current BeamNG.tech program. Error occurs with BeamNGpy v1.28 and BeamNG.tech v0.31.3.0.

aivora-beamng, 2024-04-16:
Hi, you're right and we will fix the bug in the next release of BeamNGpy. In the meantime, you can apply the patch to your copy of BeamNGpy by editing the `beamngpy/scenario/scenario_object.py` file as follows:

[image]

Thanks for the provided source code.

aivora-beamng, 2024-05-17:
Fixed in BeamNGpy v1.29, thank you for reporting.

closed 2024-05-17 by aivora-beamng