BeamNG/BeamNGpy #267: Socket message transfer very slow

Leviathan321, 2024-04-27:
I am doing research with beamng and I am using basically a driving controller which has to retrieve the camera images process and calculating driving commands (similiar to here: https://github.com/sbft-cps-tool-competition/cps-tool-competition).

However, I am facing the problem that retreiving the sensor data and as well other state information takes roughly 0.20 s (BeamNG 1.24). 

It look to me that the socket communication is very slow. Did anyone had similar issues before?

aivora-beamng, 2024-04-29:
Hi, 0.2s seems very slow indeed, are you creating multiple large-resolution camera sensors? If not, maybe you have some software installed that filters/slows the connection? If you share your script (here or to tech@beamng.gmbh), we can try to run it and test its performance at our machine.

Leviathan321, 2024-04-29:
Actually, I am directly using the code from the competition here: https://github.com/sbft-cps-tool-competition/cps-tool-competition. I have tried the code on different machines. I was running it on 
1. Windows 11, with an Nvidia Geforce GTX 4090
2. Windows 10, with an Nvidia Geforce 1080 Ti
There was a similar problem...

AbdelrahmanElsaidElsawy, 2025-12-04:
Hi @Leviathan321,

Just wanted to follow up on the BeamNGpy issues overview you requested! Thanks again for bringing it up. But we couldn't reproduce similar behavior, so we couldn't find the root cause of this issue. If you notice anything else or have suggestions for improvements, feel free to open an issue or reach out anytime.

Also, you're welcome to join the conversation on our community forum: https://forum.beamng.tech/

All the best,
Abdulrahman Saeed
Research Software Engineer
BeamNG.tech Support Team