BeamNG/BeamNGpy #274: Setting vehicle part config causes downstream ConnectionRefusedError: [WinError 10061]

garethjns, 2024-06-11:
Hi,

Not sure if this is a bug or user error, but whenever I set a part config on a vehicle then call another command on the vehicle (such as changing the AI settings) I get the following error:

````
Traceback (most recent call last):
  File "C:\envs\beamng\Lib\site-packages\beamngpy\connection\prefixed_length_socket.py", line 36, in _recv_exactly
    received = self.skt.recv(min(BUF_SIZE, length))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionAbortedError: [WinError 10053] An established connection was aborted by the software in your host machine

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\project_path\scripts\example.py", line 21, in <module>
    vehicle.ai_set_mode("manual")
  File "C:\envs\beamng\Lib\site-packages\beamngpy\api\vehicle\ai.py", line 41, in set_mode
    self._send(data).ack('AiModeSet')
  File "C:\envs\beamng\Lib\site-packages\beamngpy\connection\connection.py", line 275, in ack
    message = self.recv()
              ^^^^^^^^^^^
  File "C:\envs\beamng\Lib\site-packages\beamngpy\connection\connection.py", line 267, in recv
    message = self.connection.recv(self.req_id)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\envs\beamng\Lib\site-packages\beamngpy\connection\connection.py", line 209, in recv
    message = self.skt.recv()
              ^^^^^^^^^^^^^^^
  File "C:\envs\beamng\Lib\site-packages\beamngpy\connection\prefixed_length_socket.py", line 64, in recv
    packed_length = self._recv_exactly(self.HEADER_BYTES)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File C:\envs\beamng\Lib\site-packages\beamngpy\connection\prefixed_length_socket.py", line 38, in _recv_exactly
    self.reconnect()
  File "C:\envs\beamng\Lib\site-packages\beamngpy\connection\prefixed_length_socket.py", line 83, in reconnect
    self.skt.connect((self.host, self.port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
````

This example reproduces the behavior:

````python
from beamngpy import BeamNGpy, Scenario, Vehicle

beamng = BeamNGpy(
        host='localhost', port=64256,
        home='C:\Program Files (x86)\Steam\steamapps\common\BeamNG.drive')
beamng.open(launch=True)
scenario = Scenario('west_coast_usa', 'ai_sine')
vehicle = Vehicle('ego_vehicle', model='etk800', license='AI')
scenario.add_vehicle(vehicle, pos=(-769.1, 400.8, 142.8), rot_quat=(0, 0, 1, 0))
scenario.make(beamng)
beamng.scenario.load(scenario)
beamng.scenario.start()

# Setting a part config respawns vehicle as expected
vehicle.set_part_config(vehicle.get_part_config())

# vehicle.disconnect()
# vehicle.connect(bng=beamng)

# Error occurs here unless disconnect/connect lines are uncommented above
vehicle.ai_set_mode("manual")
````

If the lines
````python
# vehicle.disconnect()
# vehicle.connect(bng=beamng)
````
are uncommented above, the error does not occur. Reconnecting to the vehicle after setting the part config wasn't required in older versions of `beamngpy`.

Tested with:
Windows 11
Python 3.9 and 3.11 (managed by conda)
BeamNG 0.32
`beamnypy==1.29`

AbdelrahmanElsaidElsawy, 2024-08-29:
Hello @garethjns,

Thank you for reporting the issue with the ```set_part_config``` function in BeamNGpy. We appreciate your detailed explanation and the example you provided. We will investigate this issue and work on a fix as soon as possible.

aivora-beamng, 2024-10-07:
Hi, this is fixed in BeamNGpy v1.30, which has just been released. Thank you for the report.

closed 2024-10-07 by aivora-beamng