BeamNG/BeamNGpy #275: Running BeamNG remotely: Error when running BeamNG via Remote Desktop Control Connection, no start with SSH

Leviathan321, 2024-06-16:
I way trying to run BeamNG (py versions 0.26.2 and 0.23.5) remotely via Windows Remote Desktop Connection. I am successfull in starting some scenario with BeamNG, however, when I close the RDC I get the following error. This has been tried also on different devices. Running without RDC works.

[image]

I was trying to run BeamNG via SSH, here the scenario does not start without error message.

aivora-beamng, 2024-09-11:
Hi, I do not run into this issue (using Microsoft Remote Desktop on macOS), it may be caused by your remote desktop client (which one do you use?).

This message also suggests there is no GPU available to use for BeamNG, maybe try to override Windows options that specify the GPU in Remote Desktop connections: https://answers.microsoft.com/en-us/windows/forum/all/use-host-gpu-in-remote-desktop-session/34e1fd21-bf52-4f73-9907-3ce7359546d8

Leviathan321, 2024-09-11:
Thanks for the comment, I am using Microsoft Remote Desktop Connect from Windows.