BeamNG/BeamNGpy #277: The BeamNG.tech License limitation

yangqu, 2024-07-29:
Hi,I test the advanced api like sensors but I can't use it and the error is now BeamNG.tech License.How can I solve the problem to do the research

AbdelrahmanElsaidElsawy, 2024-07-30:
Hello, @yangqu,

To resolve the issue with the BeamNG.tech license, please follow these steps to obtain the necessary license:

Eligibility to Apply:

You must be currently enrolled in a college, program, course, school, class, etc., or
You must be currently working as academic staff in a research institute, college, school, etc. (e.g., professors, lecturers, research assistants).
Ensure that your usage and content comply with the BeamNG Terms & Conditions in the Academic License Agreement (which will be sent after your application is approved).
How to Apply:

Submit your application through [register.beamng.tech](https://register.beamng.tech/) using your professional/academic email address as proof of eligibility.
Provide comprehensive details about your research/project/use case.
Please let us know if you have any other questions or need further assistance.

Best regards,
Abdulrahman Saeed
Research Software Engineer
BeamNG Team

AbdelrahmanElsaidElsawy, 2024-08-28:
Hi,

As there hasn’t been any further activity on this issue, I’m going to close the ticket for now. If you encounter any more problems or have additional questions, feel free to reopen the issue or start a new one. We’re here to help!

Thank you for using BeamNGpy.

Best regards,
Abdelrahman Elsaid

closed 2024-08-28 by AbdelrahmanElsaidElsawy