BeamNG/BeamNGpy #280: How to quickly determine the location coordinates where a vehicle can be placed on a map？

lsm2842035890, 2024-08-31:
scenario = Scenario('west_coast_usa', 'scenario_1')
    vehicle_1 = Vehicle('vehicle_1', model='etk800', licence='vehicle_1')
    scenario.add_vehicle(vehicle_1, pos=(-717, 101, 118.5), rot_quat=(0, 0, 0.92, -0.38))
①How do I get all the possible points on this map where I can place a vehicle? When I load a new map, I don't know how to determine a coordinate where I can place a vehicle.
②What is the relationship between the direction in the map and rot_quat?

aivora-beamng, 2024-09-11:
Hi, there are a few ways how to determine the coordinates:
1. Open the simulator console (`~`), switch to the vehicle that you want to get the position/rotation of, and then write the following code into the console:
```lua
dump(obj:getPosition())
dump(obj:getRotation())
```
[image]

2. Open the World Editor (F11), click on the vehicle you want to get the position of. Then you can see its position and rotation in Euler degrees from the right-side panel.
[image]

3. Press Alt+O to capture a screenshot and upload it to BeamNG media server. Then you can get the position/rotation of the **camera** by checking the information pop-up on the website. Try with this example: https://media-cdn.beamng.com/m/ecNz3OpS78oob1JU

[image]

aivora-beamng, 2024-10-07:
Hi, closing this as the answer is provided and there is no activity. Feel free to reopen if you have more questions.

closed 2024-10-07 by aivora-beamng