BeamNG/BeamNGpy #282: Running BeamNG headless

Leviathan321, 2024-09-09:
Is there a way to run BeamNG headless? I would need to run it via ssh on a remote machine without using the GUI. When I run it via ssh it does not produce an error but it also does not run.

aivora-beamng, 2024-09-11:
Hi, the way to run BeamNG headless is to run it with these command-line arguments:
```bash
BinLinux/BeamNG.tech.x64.exe -gfx null -batch
```

Unfortunately, the feature is still in early development, so this is not documented and you will run into issues trying to use this headless version of BeamNG. We will update you here when the feature is ready to be used.

Leviathan321, 2024-09-22:
Thank you. Do you also know if there is way to run BeamNG via ssh? If I want to run it via ssh, the simulator is not opened.

aivora-beamng, 2024-11-12:
Hi, to run BeamNG via ssh, you can either:
1) Use X server forwarding (`ssh -X` or `ssh -Y`)
```bash
ssh -X user@server
BinLinux/BeamNG.tech.x64 <arguments>
```

2) Use the X virtual framebuffer on the server
```bash
sudo apt-get install xvfb
Xvfb :99 # open a X virtual framebuffer on the display no. 99
```
and from another terminal instance you can run BeamNG:
```bash
DISPLAY=:99 BinLinux/BeamNG.tech.x64 <arguments>
```

You may encounter some bugs with `Xvfb`, we are actively working to make the experience smoother for next updates of BeamNG.tech.

aivora-beamng, 2024-12-11:
Hi, we released the instructions for running with Xvfb headless mode (also usable for SSH) in our documentation: https://documentation.beamng.com/beamng_tech/arguments_and_settings/#headless-mode

Tested on BeamNG.tech 0.34 and above.

The-Real-Thisas, 2025-03-16:
@aivora-beamng  Any idea when the `-headless` will be properly supported and implemented into beamng py at least for linux. 

I'm actively working on getting a model to complete https://github.com/BeamNG/BeamNG.gym and training takes a very long time. 

Ideally I want to modify https://github.com/BeamNG/BeamNG.gym to run multiple truly headless instances of BeamNG.tech running faster than realtime.

The-Real-Thisas, 2025-03-16:
Willing to create a pull just need to know whats needed for it at this point because I have no control over BeamNG.tech so when `--headless` is implemented on that end I can help implement it into BeamNGpy and then into BeamNG.gym

aivora-beamng, 2025-03-17:
Hi @The-Real-Thisas, we release the initial implementation of the `-headless` argument in BeamNG.tech 0.36, as with other command-line arguments you can use Python to start BeamNG with them like so:
```python
beamng = BeamNGpy('localhost', 25252)
beamng.open(None, '-headless')
```

Not sure if performance improvements will be huge, we are still doing full render pass in headless mode (needed for example for camera data), only presentation is skipped.

You can get some information about running faster than realtime from Python here: https://forum.beamng.tech/t/simulation-time-speed/124

aivora-beamng, 2025-04-09:
Hi, we fully released headless mode in BeamNG.tech 0.35, see documentation for more details: https://documentation.beamng.com/beamng_tech/headless_mode/

Closing the issue.

closed 2025-04-09 by aivora-beamng