BeamNG/BeamNGpy #284: The annotation result of a camera sensor does not update when we attach a Lidar sensor simultaneously

dli2016, 2024-09-26:
We attach a camera and a Lidar for the vehicle. 
`
lidar = Lidar('lidar1', beamng, vehicle, pos=(0, 0, 1.7), requested_update_time=0.05, 
                   is_using_shared_memory=False, is_visualised=True, frequency=3,
                   is_360_mode=False, horizontal_angle=120, max_distance=85)
front_camera = Camera('front_camera', beamng, vehicle,
        pos=(-0.3, -2.1, 1), dir=(0, 0, 0), field_of_view_y=120, resolution=(256, 256),
        is_render_colours=True, is_render_depth=False, is_render_annotations=True, 
        is_render_instance=False)
`
However, the annotation result of the camera does not update when we save the data as follows:
`front_cam_data = front_camera.poll()
 annotation = front_cam_data['annotation'].convert('RGB')
 annotation.save('xxx.jpg')
`
We always get the annotation result of the first frame. The color image and the depth image are correct. And we find that the function get_full_poll_request() must be used to obtain the correct annotation result. Both poll() and stream() are performed abnormally.
Could you please help us check if we use the api correctly? Thank you!

aivora-beamng, 2024-09-26:
Hi, thank you for the report. This is a bug and we managed to reproduce with your code, we are now working on a fix.

dli2016, 2024-09-27:
Thank you for the quick reply and we look forward for the new version.

dli2016, 2024-12-26:
This bug has been fixed with beamngpy 1.31 and beamng 0.34.2. Thanks again!

closed 2024-12-26 by dli2016