BeamNG/BeamNGpy #285: Recording ROS2 bags of LiDAR and IMU messages

ZHDphilip, 2024-11-25:
Hi, I am trying to use BeamNG to collect ros2 bagfiles which will be postprocessed for model training. I set up my LiDAR and IMU sensor as follow:

```
lidar = Lidar(
    ...
    requested_update_time = 0.05,
    frequency = 20,
    is_using_shared_memory=True,
    is_streaming = True
)

imu = AdvancedIMU(
    ...
    gfx_update_time = 0.01,
    physics_update_time = 0.01,
    is_send_immediately = True
)
```
I saw that the examples use an infinite `while` loop with `sleep()` to simulate polling in readings at desired frequency, however it is single threaded and thus non-optimal if I want to record them at exactly 20Hz and 100Hz, respectively. Is there a way to achieve this? For example, setting up some kind of trigger, such that beamng publishes a ros2 pointcloud/imu message as soon as they are collected?

aivora-beamng, 2024-11-29:
Hello, we do not support this at the moment unfortunately, however we may try to implement support for set frequency sensor publishing in the future. You should be able to use two threads to poll for the data independently.

ZHDphilip, 2024-12-01:
I see. Thanks for the response, I will try polling data using separate threads as you suggested.

closed 2024-12-01 by ZHDphilip