BeamNG/BeamNGpy #286: No effect of beamngpy.Scenario.add_checkpoints() and similar in usage after beamngpy.Scenario.load()

omeier919, 2024-12-16:
The beamngpy.Scenario.add_checkpoints() does not have any effect if used after loading a scenario. This is _not_ documented in the [BeamNGpy Reference](https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.Scenario.add_checkpoints), nor is a Warning issued at runtime.

Minimal reproducable Code (modified from examples/checkpoints.py):

```
from beamngpy import BeamNGpy, Scenario, Vehicle

if __name__ == "__main__":
    with BeamNGpy("localhost", 25252) as bng:
        scenario = Scenario("tech_ground", "waypoint demo")
        vehicle = Vehicle("test_car", model="etk800", rot_quat=(0, 0, 0, 1))
        scenario.add_vehicle(vehicle)

        positions = [(0, -10, 0), (0, -20, 0), (0, -30, 0)]
        scales = [(1.0, 1.0, 1.0)] * 3
        scenario.make(bng)

        bng.scenario.load(scenario)
        scenario.add_checkpoints(positions, scales)
        bng.scenario.start()
        input("Press Enter to exit...")
```

**Expected Behavior**:
Checkpoint is added, or RuntimeWarning or similar is thrown.
**Actual Behavior**:
No checkpoint is added (can be confirmed by checking result of scenario.find_waypoints()). No Notice about not adding the checkpoint is given.

omeisosis, 2024-12-16:
Apologies if this behavior should have been obvious or is known or even intended, i thought i would submit an issue just in case, because i found it unintuitive. I personally require no further assistance.

aivora-beamng, 2025-01-02:
Hi, thank you for the report! We will add information about it in the reference documentation and see whether we can implement this for the future releases.