BeamNG/BeamNGpy #287: Multiple cameras capture images in parallel

YaoningLi, 2025-01-02:
My beamng.tech verison: 0.33, beamngpy version: 1.30
I loaded a custom map scene and placed 4 fixed cameras in the scene. The code is as follows:
Load map:
```
bng = BeamNGpy('localhost', 64256, home='my_path\BeamNG\BeamNG.tech.v0.33.3.0', user='my_path')
bng.open(debug=True)
    #Load Scenario
scenario = Scenario("fresh", "fresh")
```
Set cameras:
```
 for camera_id in cameras_ids:  
        camera_name = 'camera{}'.format(camera_id+1)  
        camera_param = cfg[camera_name]  
        camera_pos = camera_param['pos']  
        camera = Camera(camera_name, bng,pos=camera_pos, **custom_params) 
        cameras[camera_name] = camera
```

My goal is to drive a vehicle through different cameras in the scene, and I need four cameras to capture images in parallel to keep the capture time consistent. Therefore, I tried to use mutiprocess to create four processes, each process accepts a different camera instance, and calls the poll() method in the process to capture the picture. However, when I run the code, it tells me that the camera cannot be serialized and therefore cannot be passed into the process. I tried to change this to only pass in the position parameters of different cameras and instantiate a camera in each process. The code is as follows:
```
for camera_id in camera_ids:
        process = multiprocessing.Process(target=run, args=(cfg,camera_id,bng,))
        processes.append(process)
        process.start() 
    for process in processes:
        process.join()
```

```
def run(cfg,camera_id,bng):
    # print(cameras_param['resolution'])
    camera_params = cfg['cameras']
    bng, camera_name,camera = initialize_camera(camera_params,bng,camera_id)
```
But it seems to be stuck at the point where the camera is initialized.
I noticed the multishot example, but I don't think it's really parallelized because you still have to iterate over each camera position to capture the image.
So, How can I achieve true multi-camera parallel capture?

aivora-beamng, 2025-01-06:
Hi, it's not optimal to use the multiprocessing module in this case, as the cameras will still be a part of the same simulator process. If you need to poll the cameras at the same time instance, you can use the stepping feature of the simulator:
```python
from beamngpy import BeamNGpy, Scenario
from beamngpy.sensors import Camera
from beamngpy.vehicle.vehicle import Vehicle
import matplotlib.pyplot as plt

cfg = {
    "camera1": {
        "pos": [-1.0, -1.0, 1.0],
    },
    "camera2": {
        "pos": [1.0, 1.0, 1.0],
    },
    "camera3": {
        "pos": [-1.0, 1.0, 1.0],
    },
    "camera4": {
        "pos": [1.0, -1.0, 1.0],
    },
}

SPS = 60

if __name__ == "__main__":
    bng = BeamNGpy("localhost", 25252).open()
    scenario = Scenario("tech_ground", "fresh")
    vehicle = Vehicle("ego", "etk800")

    scenario.add_vehicle(vehicle)
    scenario.make(bng)
    bng.settings.set_deterministic(SPS)
    bng.control.pause()
    bng.scenario.load(scenario)
    bng.scenario.start()

    camera_ids = [0, 1, 2, 3]
    cameras = {}
    for camid in camera_ids:
        cam_name = f"camera{camid + 1}"
        cameras[cam_name] = Camera(cam_name, bng=bng, vehicle=vehicle, **cfg[cam_name])

    passed_time = 0
    bng.step(SPS)
    while True:
        print(f"Timestamp {passed_time:.2f}s")
        readings = {
            cam_name: camera.poll()["colour"].convert("RGB")
            for cam_name, camera in cameras.items()
        }
        f, axarr = plt.subplots(2, 2)
        axarr[0, 0].imshow(readings["camera1"])
        axarr[0, 1].imshow(readings["camera2"])
        axarr[1, 0].imshow(readings["camera3"])
        axarr[1, 1].imshow(readings["camera4"])
        plt.show()

        wait_steps = 5 * SPS  # waiting 5 seconds
        bng.step(wait_steps)
        passed_time += wait_steps / SPS
```

YaoningLi, 2025-01-08:
Thank you for your response! I understand that the key to your method is using  `bng.control.pause()` and bng.control.step() to achieve simultaneous polling and image capturing. Can I understand it this way: calling `bng.control.pause()` will block the BeamNG simulator, and the simulation will only proceed when `bng.control.step()` is called? In the while loop, each iteration will simulate for a duration of wait_steps?

Additionally, I have another question: does `bng.control.pause()` also cause the frontend visuals to freeze? For example, if I’m driving a car, would `bng.control.pause()` also stop the car from moving? If the car continues to move during the BeamNG block period, will the vehicle's movement data be captured when `bng.control.step()’`is called later? In other words, will the data capture only be delayed relative to the frontend visuals, without any data loss?

Thank you again for your help！

YaoningLi, 2025-01-13:
Thank you again for answering my questions！

closed 2025-01-13 by YaoningLi