BeamNG/BeamNGpy #288: How can I get the lane info of a decal road?

xTayEx, 2025-01-08:
Hi BeamNG team! First of all, thank you for creating such a wonderful simulator and developer-friendly python API. I'm currently    
working with the `beamngpy` library to control in-game vehicles.

My goal is to have a vehicle perform a lane change and verify that the action has been successfully executed. To do this, I would like to    
compare the vehicle's coordinates with those of the target lane. After thoroughly reviewing the documentation, I was unable to find a        
suitable method or property that provides the necessary information for this comparison. Could you kindly provide me some guidance?

Thank you!

cpap-beamng, 2025-01-10:
Thank you for your kind words. 
So, as much as I understand your application, let me recommend here few sensors, which you may find documented at [this page](https://beamngpy.readthedocs.io/en/latest/beamngpy.html#id2) . These sensors would be the Roads sensor, where you can use signals like the distances to the center line, the left and the right road edges, as well as the halfwidth and number of lanes signals. Further, you can consider also the State sensor and the GPS, for the vehicle's position. 
Hope this helps.

xTayEx, 2025-01-12:
> Thank you for your kind words. So, as much as I understand your application, let me recommend here few sensors, which you may find documented at [this [...]

Thanks for your reply! I gave it a try, and it works for what I need.

xTayEx, 2025-01-12:
@cpap-beamng Another question is, how can I specify the lane of number of a decal road? In the [doc](https://beamngpy.readthedocs.io/en/latest/beamngpy.html#roads), I couldn't find any related arguments.

closed 2025-12-04 by cpap-beamng