BeamNG/BeamNGpy #289: if RoadsSensor then BeamNG.tech freeze

dsm, 2025-01-14:
I use below code to test my RL test but if i used `RoadsSensor` then freeze about 1000 iterations but I remove then no problem but I need RoadSensor. Btw normally it runs 30it/s but enable `RoadsSensor` decresed to 20it/s

```py
self.road_sensor = RoadsSensor("rs1", self.bng, self.vehicle,physics_update_time=1)

def _compute_reward(self, speed, current_location, camera_image):
        reward = 0.0
        # Poll the road sensor for data
        rs1 = self.road_sensor.poll()

        if len(rs1) != 0:  # Ensure the sensor data is available
            # 1. Lane Keeping Reward
            dist_to_center = rs1[0.0].get("dist2CL", None)
            if dist_to_center is not None:
                reward += max(
                    0, 1 - abs(dist_to_center)
                )  # Closer to centerline gets higher reward

            # 2. Staying on Road Penalty
            dist_to_left = rs1[0.0].get("dist2Left", None)
            dist_to_right = rs1[0.0].get("dist2Right", None)
            half_width = rs1[0.0].get("halfWidth", None)
            if (
                dist_to_left is not None
                and dist_to_right is not None
                and half_width is not None
            ):
                if dist_to_left < 0 or dist_to_right < 0:  # Vehicle is off the road
                    reward -= 10
                elif abs(dist_to_left - dist_to_right) > half_width:  # Veering too much
                    reward -= 5

            # 4. Curvature and Heading Alignment Reward
            road_radius = rs1[0.0].get("roadRadius", None)
            heading_angle = rs1[0.0].get("headingAngle", None)
            if heading_angle is not None:
                heading_angle = heading_angle * 180 / np.pi  # Convert to degrees
            if road_radius is not None and not np.isnan(road_radius):
                reward += 1 / (
                    1 + abs(heading_angle)
                )  # Smaller heading angle gets higher reward

            # 6. Alignment with Centerline
            x_closest = rs1[0.0].get("xP0onCL", None)
            y_closest = rs1[0.0].get("yP0onCL", None)
            if x_closest is not None and y_closest is not None:
                vehicle_x, vehicle_y = current_location[:2]
                distance_to_closest = np.sqrt(
                    (vehicle_x - x_closest) ** 2 + (vehicle_y - y_closest) ** 2
                )
                reward += max(
                    0, 1 - distance_to_closest
                )  # Higher reward for staying close to centerline

        # Damage Penalty
        if self.damage_sensor["damage"] > 0:
            self.obstacles_hit += 1
            reward -= 50.0  # Large penalty for damage

        self.last_location = current_location
        return reward
```

dsm, 2025-01-14:
I tried BeamNG.tech.v0.33.3.0 and then this build not freezed.

cpap-beamng, 2025-01-27:
Thanks for the report ! Under investigation...