BeamNG/BeamNGpy #290: Unable to create Johnson Valley scenario on BeamNG.tech 0.34.2.0

raymondsong00, 2025-01-21:
Unable to load johnson_valley scenario with BeamNG.tech 0.34.2.0.

Version: BeamNG.tech 0.34.2.0, beamngpy 1.31
OS: Ubuntu 22.04 and Windows 10

Code (Modified from Readme):
```
from beamngpy import BeamNGpy, Scenario, Vehicle

if __name__ == "__main__":
    bng = BeamNGpy("localhost", 25252)
    bng.open()
    scenario = Scenario("johnson_valley", "example") 
    vehicle = Vehicle("ego_vehicle", model="etk800", license="PYTHON")

    scenario.add_vehicle(vehicle, rot_quat=(0, 0, 0, 1))
    scenario.make(bng)

    bng.scenario.load(scenario)
    bng.scenario.start()
    input("Hit Enter when done...")
    bng.disconnect()
```

Expected Behavior:
Create new Johnson Valley Scenario with vehicle and load in BeamNG.tech v0.34.2.0.

Actual Behavior:
Error message: `beamngpy.logging.BNGValueError: Level not found: "/levels/johnson_valley/main.level.json"`

Additional Context:
This code works as intended on 
- BeamNG.tech 0.33.3.0, BeamNGpy 1.30, Ubuntu 22.04
- BeamNG.tech 0.31.3.0, BeamNGpy 1.28, Ubuntu 22.04
- BeamNG.tech 0.31.3.0, BeamNGpy 1.28, Windows 10

Tried 0.34.2.0 + Ubuntu with a clean download and userpath with the same issue.

Loading scenarios from Johnson Valley with `bng.scenario.get_scenarios()['johnson_valley']` works as intended.

aivora-beamng, 2025-01-21:
Hi @raymondsong00,
thank you very much for the detailed report! There's a bug in the `load_scenario` implementation, we will try to fix it ASAP on the upstream version. In the meantime, you can use this patched version of `techCore.lua` for BeamNG.tech 0.34.2.0, which you can copy over to `%BNG_HOME%/lua/ge/extensions/tech`:

[techCore.lua.zip](https://github.com/user-attachments/files/18489019/techCore.lua.zip)

raymondsong00, 2025-01-21:
Thanks for the quick fix! Worked on both Windows 10 and Ubuntu 22.04.

I will leave the issue open for now until the upstream version is fixed.

aivora-beamng, 2025-04-09:
Hi, fixed in BeamNGpy 1.32 (BeamNG.tech 0.35), closing.

closed 2025-04-09 by aivora-beamng