BeamNG/BeamNGpy #293: Weird Acceleration Values

alfredgu001324, 2025-02-25:
Hi Team,

Thanks for the great contributions! I am currently trying out the IMU sensor, but found that the acceleration values are quite large (like 30/40 m/s^2), which I think is abnormal. I am wondering what is the unit of it? Thanks! I have attached my code below:

```
def track_2():
    # set_up_simple_logging()
    beamng = BeamNGpy('localhost', 25252, home="/home/guxunjia/masters_project/BeamNG.tech.v0.34.2.0/")
    beamng.open()

    scenario = Scenario("smallgrid", "Oval_Track")
    vehicle = Vehicle("ego_vehicle", model="scintilla", license="Aoligei")

    orig = (0, 0, 0)

    scenario.add_vehicle(vehicle, pos=orig, rot_quat=(0, 0, 1, 0))
    scenario.make(beamng)

    beamng.control.pause()
    beamng.settings.set_deterministic(60)

    points = []
    script = []
    point_color = (1, 0, 0, 0.5)
    sphere_coordinates = []
    sphere_radii = []
    sphere_colors = []

    straight_length = 400
    radius = 100
    arc_points = 100

    points = []

    for i in np.linspace(0, straight_length, 100):
        points.append((0, i, 0))

    for angle in np.linspace(np.pi, 0, arc_points):
        x = radius * np.cos(angle) + radius
        y = straight_length + radius * np.sin(angle)
        points.append((x, y, 0))

    for i in np.linspace(straight_length, 0, 100):
        points.append((2 * radius, i, 0))

    for angle in np.linspace(0, np.pi, arc_points):
        x = radius * np.cos(angle) + radius
        y = - radius * np.sin(angle)
        points.append((x, y, 0))

    for idx, point in enumerate(points):
        script.append({"x": point[0], "y": point[1], "z": point[2], "t": idx * 0.1})
        if idx % 10 == 0:
            sphere_coordinates.append(point)
            sphere_radii.append(0.5)
            sphere_colors.append((1, 0, 0, 0.8))

    beamng.scenario.load(scenario)
    beamng.scenario.start()

    imu = AdvancedIMU("accel1", beamng, vehicle, dir=(1,0,0), up=(0,0,1), is_send_immediately=True)
    electrics = Electrics()
    timer = Timer()
    vehicle.sensors.attach("electrics", electrics)
    vehicle.sensors.attach("timer", timer)
    beamng.debug.add_polyline(points, point_color, cling=True, offset=0.1)
    beamng.debug.add_spheres(sphere_coordinates, sphere_radii, sphere_colors, cling=True, offset=0.1)

    vehicle.ai.set_script(script)

    beamng.control.resume()
    data_list = []
    
    # Alternative approach using simulation time
    dt = 0.04  # 25Hz sampling rate
    vehicle.sensors.poll()
    next_sample_time = vehicle.sensors["timer"]["time"]
    
    print("Collecting data at 25Hz. Press Ctrl+C to stop...")
    try:
        while True:
            vehicle.sensors.poll()
            current_sim_time = vehicle.sensors["timer"]["time"]
            
            if current_sim_time >= next_sample_time:
                imu_data = imu.poll()
                sensors = vehicle.sensors
                state = vehicle.state
                
                # Calculate yaw angle (phi) from direction vector
                phi = np.arctan2(state["dir"][1], state["dir"][0])
                
                # Transform velocity from global to local coordinates
                vx_global = state["vel"][0]
                vy_global = state["vel"][1]
                
                # Transform acceleration from global to local coordinates
                ax_local = imu_data["accSmooth"][0]
                ay_local = imu_data["accSmooth"][1]
                az_local = imu_data["accSmooth"][2]

                print(f'ax_local: {ax_local}, ay_local: {ay_local}, az_local: {az_local}')
                print(f'Angular Velocity: {imu_data["angVelSmooth"][2]}')
                
                # Rotation matrix for local coordinates
                vx_local = vx_global * np.cos(-phi) - vy_global * np.sin(-phi)
                vy_local = vx_global * np.sin(-phi) + vy_global * np.cos(-phi)

                # Prepare data for each timestamp
                data_row = {
                    'time(s)': sensors["timer"]["time"],
                    'x(m)': state["pos"][0],
                    'y(m)': state["pos"][1],
                    'vx(m/s)': vx_local,
                    'vy(m/s)': vy_local,
                    'phi(rad)': phi,
                    'delta(rad)': np.radians(sensors["electrics"]["steering"]),
                    'omega(rad/s)': imu_data["angVelSmooth"][2],
                    'ax(m/s^2)': 0,
                    'deltadelta(rad/s)': 0,
                    'wheel_fl(kmph)': sensors["electrics"]["wheelspeed"] * 3.6,
                    'wheel_fr(kmph)': sensors["electrics"]["wheelspeed"] * 3.6,
                    'wheel_rl(kmph)': sensors["electrics"]["wheelspeed"] * 3.6,
                    'wheel_rr(kmph)': sensors["electrics"]["wheelspeed"] * 3.6,
                    'roll(rad)': state["dir"][0],
                    'throttle_ped_cmd(%)': sensors["electrics"]["throttle"] * 100,
                    'brake_ped_cmd(kPa)': sensors["electrics"]["brake"] * 2757.89990234
                }
                data_list.append(data_row)
                
                # Schedule next sample
                next_sample_time += dt
                
            # Small sleep to prevent CPU overuse
            time.sleep(0.001)
            
    except KeyboardInterrupt:
        print("\nData collection stopped by user")
        
    finally:
        # Save data and cleanup
        beamng.disconnect()
        
        # Save to CSV
        csv_file = 'vehicle_states.csv'
        with open(csv_file, 'w', newline='') as f:
            writer = csv.DictWriter(f, fieldnames=data_list[0].keys())
            writer.writeheader()
            writer.writerows(data_list)

    return data_list
```

The output is:
```
ax_local: -36.17008145362029, ay_local: 14.673618256379674, az_local: 32.055790612528995
Angular Velocity: -0.3364974276358178
ax_local: -35.8545063758097, ay_local: 13.855194362467698, az_local: 35.082536558557685
Angular Velocity: -0.3229758932305078
ax_local: -45.51596230133285, ay_local: 24.837369380434453, az_local: 34.21708671691664
```

Thanks in advance for any inputs!
```

aivora-beamng, 2025-02-26:
Hi,
we are pretty confident the acceleration values are accurate. The problem here is that the default position of an IMU sensor is `(0, 0, 1.7)`, which means it is attached on an invisible rod above the vehicle. Try to set the position of the AdvancedIMU to something like `AdvancedIMU(pos=(0, 0, 0.5))`. We are discussing how to make the sensor setup easier, thank you for the input! Let us know if you're getting better results now.

alfredgu001324, 2025-02-26:
It seems that the acceleration values are still not correct. For your reproduction, here is my code:

```
import numpy as np
import matplotlib.pyplot
from beamngpy import BeamNGpy, Scenario, Vehicle
from beamngpy.sensors import AdvancedIMU, Electrics, Timer, Camera

import csv

def create_track():
    points = []
    script = []
    point_color = (1, 0, 0, 0.5)
    sphere_coordinates = []
    sphere_radii = []
    sphere_colors = []

    straight_length = 400
    radius = 100
    arc_points = 100

    points = []

    for i in np.linspace(0, straight_length, 100):
        points.append((0, i, 0))

    for angle in np.linspace(np.pi, 0, arc_points):
        x = radius * np.cos(angle) + radius
        y = straight_length + radius * np.sin(angle)
        points.append((x, y, 0))

    for i in np.linspace(straight_length, 0, 100):
        points.append((2 * radius, i, 0))

    for angle in np.linspace(0, np.pi, arc_points):
        x = radius * np.cos(angle) + radius
        y = - radius * np.sin(angle)
        points.append((x, y, 0))

    for idx, point in enumerate(points):
        script.append({"x": point[0], "y": point[1], "z": point[2], "t": idx * 0.1})
        if idx % 10 == 0:
            sphere_coordinates.append(point)
            sphere_radii.append(0.5)
            sphere_colors.append((1, 0, 0, 0.8))

    return points, script, sphere_coordinates, sphere_radii, sphere_colors, point_color

cfg = {
    "camera1": {
        "pos": [-1.0, -1.0, 1.0],
    },
    "camera2": {
        "pos": [1.0, 1.0, 1.0],
    },
    "camera3": {
        "pos": [-1.0, 1.0, 1.0],
    },
    "camera4": {
        "pos": [1.0, -1.0, 1.0],
    },
}

if __name__ == "__main__":
    beamng = BeamNGpy('localhost', 25252, home="/home/guxunjia/masters_project/BeamNG.tech.v0.34.2.0/")
    beamng.open()
    scenario = Scenario("smallgrid", "Oval_Track")
    vehicle = Vehicle("ego_vehicle", model="scintilla", license="Aoligei")
    
[... 101 lines cut ...]
                'throttle_ped_cmd(%)': sensors["electrics"]["throttle"] * 100,
                'brake_ped_cmd(kPa)': sensors["electrics"]["brake"] * 2757.89990234
            }
            data_list.append(data_row)
            
            # Step simulation forward by 1 frame (at 25Hz, this is 0.04s)
            beamng.step(1)
            passed_time += 1/SPS
            
    except KeyboardInterrupt:
        print("\nData collection stopped by user")
        
    finally:
        # Save data and cleanup
        beamng.disconnect()
        
        # Save to CSV
        csv_file = 'vehicle_states.csv'
        with open(csv_file, 'w', newline='') as f:
            writer = csv.DictWriter(f, fieldnames=data_list[0].keys())
            writer.writeheader()
            writer.writerows(data_list)
```

You can also use this to check the time reading issue that I mentioned in another [Issue](https://github.com/BeamNG/BeamNGpy/issues/294#issuecomment-2684413039). Thank you so much for your help!

alfredgu001324, 2025-02-26:
The readings are like:
```
Timestamp 3.08s
Timer Time: 3.0400002151727676
State Time: 7.80000037048012
IMU Time: 7.796500370313879
ax_local: 41.08743538476668, ay_local: 25.27696325312417, az_local: -18.641400431587044
Angular Velocity: 0.004651483661348903
Steering: -0.41835971338187705
Vy: -0.0015905196162680122
Timestamp 3.12s
Timer Time: 3.0800002180039883
State Time: 7.840000372380018
IMU Time: 7.838500372308772
ax_local: 13.049982013416447, ay_local: 31.6694941447146, az_local: -18.850361424592318
Angular Velocity: 0.0007423753577947608
Steering: -0.33228997569454344
Vy: -0.011911682038137054
Timestamp 3.16s
Timer Time: 3.120000220835209
State Time: 7.880000374279916
IMU Time: 7.870000373804942
ax_local: -7.084058824255422, ay_local: 17.392220151304052, az_local: -14.406597838159094
Angular Velocity: 7.524012338660222e-05
Steering: -0.15276296505037854
Vy: -0.005325699641996161
Timestamp 3.20s
Timer Time: 3.1600002236664295
State Time: 7.920000376179814
IMU Time: 7.912000375799835
ax_local: 20.98104533053254, ay_local: 22.206254451803375, az_local: -21.95692076638403
Angular Velocity: -0.006618034119178521
Steering: -0.34243155985183493
Vy: -0.022740746472409954
```

dstark481, 2025-02-27:
Hello Alfred.

The IMU sensor can be very sensitive and runs at up to 2kHz - that means its highly sensitive to rapid acceleration changes and is in the range of vehicle vibration modes, so can capture their behaviour (it should be able to sample anything up to 1kHz accurately, but higher things may be in there as distortion).  There are lots of reasons why you might get noise-related spikes in acceleration readings, such as vehicle-induced mechanical vibration, road surface irregularity, sensor positioning.  If you don't want to see as many spikes, you can increase the smoothing done on the readings (this is the accel_frequency_cutoff parameter).  Try increasing it quite significantly, and this will filter out high-frequency noise from the data.  It might help to plot the data over time for all three axes and look for any patterns.  You could try some other quick things, such as switching off the vehicle engine and seeing if the readings calm down, strongly hinting at vibrational causes.

We have tested this sensor against real-world data from actual IMU sensors placed on actual cars, and it compares very favourably across a range of standard tests.  It has also been used in some peer-reviewed published work.

Best,
Dave

alfredgu001324, 2025-02-28:
I see, thanks for the reply! I will test it out and let you know if I have further questions. Thanks!

closed 2025-02-28 by alfredgu001324