BeamNG/BeamNGpy #294: Time Reading and Sensor Data Retrieval (Varying Time Interval)

alfredgu001324, 2025-02-25:
Hi Team,

Thanks for the great work! During simulation, I noticed that the reading from different time measurements are slightly different, specifically:

```
vehicle.sensors["timer"]["time"]
vehicle.state["time"]
vehicle.sensors['timer']['time']
```

I am wondering which one I should follow? 

Also typically for polling out sensor data, we use a for loop with time.sleep(). I am wondering if there is another more stable way to make sure the sensor reading intervals are constant (for example, 0.04s). Currently this for loops with time.sleep as in the example folders will give readings at varying time intervals. I am wondering if there are any instructions/examples for this? Thank you so much in advance!

aivora-beamng, 2025-02-25:
Hi,
`vehicle.sensors["timer"]["time"]` gives you the time since the scenario was started (you can check the `sensors.Timer` function in `%BNG_HOME%/lua/ge/extensions/tech/techCore.lua` for the implementation).

`vehicle.state["time"]` gives you the Lua value `obj:getSimTime()`, which is monotonically increasing and does not reset on scenario load.

To get sensor data in predefined intervals, the best way is to pause the physics and use the stepping feature. The documentation there is limited (sorry!), but you should be able to use this code to get an idea: https://github.com/BeamNG/BeamNGpy/issues/287#issuecomment-2573165063

alfredgu001324, 2025-02-25:
Thanks for the quick reply! For the time calculation, I saw the the `AdvancedIMU` also has a time reading, how does that differ from the above two?

The stepping feature is currently working as desired, now my sensor readings can come in in predefined intervals. But the simulation becomes significantly slower, I am wondering if there is any remedy for this? Thank you!

As a reference, this is my code:
```
import time
import numpy as np
import matplotlib.pyplot as plt
from beamngpy import BeamNGpy, Scenario, Vehicle, set_up_simple_logging
from beamngpy.sensors import AdvancedIMU, Electrics, Timer, Camera

import csv

def create_track():
    points = []
    script = []
    point_color = (1, 0, 0, 0.5)
    sphere_coordinates = []
    sphere_radii = []
    sphere_colors = []

    straight_length = 400
    radius = 100
    arc_points = 100

    points = []

    for i in np.linspace(0, straight_length, 100):
        points.append((0, i, 0))

    for angle in np.linspace(np.pi, 0, arc_points):
        x = radius * np.cos(angle) + radius
        y = straight_length + radius * np.sin(angle)
        points.append((x, y, 0))

    for i in np.linspace(straight_length, 0, 100):
        points.append((2 * radius, i, 0))

    for angle in np.linspace(0, np.pi, arc_points):
        x = radius * np.cos(angle) + radius
        y = - radius * np.sin(angle)
        points.append((x, y, 0))

    for idx, point in enumerate(points):
        script.append({"x": point[0], "y": point[1], "z": point[2], "t": idx * 0.1})
        if idx % 10 == 0:
            sphere_coordinates.append(point)
            sphere_radii.append(0.5)
            sphere_colors.append((1, 0, 0, 0.8))

    return points, script, sphere_coordinates, sphere_radii, sphere_colors, point_color

cfg = {
    "camera1": {
        "pos": [-1.0, -1.0, 1.0],
    },
    "camera2": {
        "pos": [1.0, 1.0, 1.0],
    },
    "camera3": {
        "pos": [-1.0, 1.0, 1.0],
    },
    "camera4": {
        "pos": [1.0, -1.0, 1.0],
    },
}

if __name__ == "__main__":
    beamng = BeamNGpy('localhost', 25252, home="/home/guxunjia/masters_project/BeamNG.tech.v0.34.2.0/")
[... 98 lines cut ...]
                'wheel_rr(kmph)': sensors["electrics"]["wheelspeed"] * 3.6,
                'roll(rad)': state["dir"][0],
                'throttle_ped_cmd(%)': sensors["electrics"]["throttle"] * 100,
                'brake_ped_cmd(kPa)': sensors["electrics"]["brake"] * 2757.89990234
            }
            data_list.append(data_row)
            
            # Step simulation forward by 1 frame (at 25Hz, this is 0.04s)
            beamng.step(1)
            passed_time += 1/SPS
            
    except KeyboardInterrupt:
        print("\nData collection stopped by user")
        
    finally:
        # Save data and cleanup
        beamng.disconnect()
        
        # Save to CSV
        csv_file = 'vehicle_states.csv'
        with open(csv_file, 'w', newline='') as f:
            writer = csv.DictWriter(f, fieldnames=data_list[0].keys())
            writer.writeheader()
            writer.writerows(data_list)
```

aivora-beamng, 2025-02-26:
Hi,
the `AdvancedIMU` and other sensors which return time return the same value as `vehicle.state["time"]` (the simulation time). I ran your code and I can confirm the performance is not great. Using the [protocol debug plugin](https://github.com/BeamNG/BeamNGpy/tree/master/debug) you can see that it takes much longer than 4ms to get the data you need:
[image]

You can improve the situation a bit by calling:
```python
beamng.step(1, wait=False)
```

to remove the 4ms Python blocking in the loop. Another thing you can optimize is to use shared memory and streaming for the Camera sensor:
```python
cameras[cam_name] = Camera(cam_name, bng=beamng, vehicle=vehicle,
  is_using_shared_memory=True,
  is_streaming=True,
  **cfg[cam_name])
...
...
readings = {
    cam_name: camera.stream()["colour"].convert("RGB")
    for cam_name, camera in cameras.items()
}
```

I still don't get a "smooth" simulation with these improvements, but it may help. Thanks for testing BeamNGpy out!

alfredgu001324, 2025-02-26:
Thanks for the suggestions! Using:
```
beamng.step(1, wait=False)
```
indeed speed things up quite a bit. Although the values it returns are "offset" by the correct ones, I think I can do some data cleaning to improve it. As a reference and for the community:

Without `wait=False`, it saves the correct values with predefined intervals:
```
time(s),x(m)

0.960000067949295,0.0580275421962142

1.0000000707805157,0.063812798820436

1.0400000736117363,0.06999081512913108

1.080000076442957,0.07663644244894385

1.1200000792741776,0.0841619148850441

1.1600000821053982,0.09298034105449915

1.2000000849366188,0.101006418466568
```

With `wait=False`, it almost saves the correct values with occasional wrong offset readings:
```
0.960000067949295,0.0580275421962142

1.0000000707805157,0.063812798820436

1.0000000707805157,0.06999081512913108

1.0400000736117363,0.07663644244894385

1.080000076442957,0.0841619148850441

1.1200000792741776,0.09298034105449915

1.1600000821053982,0.101006418466568
```

alfredgu001324, 2025-02-26:
But I checked the time reading of each sensor, it seems that they indeed return different values. Here is my code:
```
            # Poll sensors
            vehicle.sensors.poll()
            imu_data = imu.poll()
            sensors = vehicle.sensors
            state = vehicle.state
            # Optional: print current simulation time
            print(f"Timestamp {passed_time:.2f}s")
            print(f"Timer Time: {sensors['timer']['time']}")
            print(f'State Time: {state["time"]}')
            print(f'IMU Time: {imu_data["time"]}')
```

which prints:
```
Timestamp 8.08s
Timer Time: 8.040000569075346
State Time: 8.080000383779407
IMU Time: 8.07450038351817
Angular Velocity: 0.010460325896274748
Steering: -0.05106082721701837
Vy: -0.0073799241815957215
Timestamp 8.12s
Timer Time: 8.080000571906567
State Time: 8.120000385679305
IMU Time: 8.116500385513064
Angular Velocity: -0.0034153088730436675
Steering: -0.049840249066283177
Vy: 0.0024063735235001415
Timestamp 8.16s
Timer Time: 8.120000574737787
State Time: 8.160000387579203
IMU Time: 8.158500387507956
Angular Velocity: -0.005944088280590255
Steering: -0.05794035637669408
Vy: -0.0075404532755655364
Timestamp 8.20s
Timer Time: 8.160000577569008
State Time: 8.2000003894791
IMU Time: 8.190000389004126
Angular Velocity: 0.0005754886688735392
Steering: -0.04241640947815361
Vy: 0.002116148327692495
Timestamp 8.24s
Timer Time: 8.200000580400229
State Time: 8.240000391378999
IMU Time: 8.23200039099902
Angular Velocity: 0.0020892406599407943
Steering: -0.051570024947435726
Vy: -0.0006660018525804598
```

`Timer Time` and `State Time` are offset by 0.04, which I think maybe the way I am polling my data might not be correct? Because the 0.04 likely comes from my set interval. But the `IMU Time` seems quite random.

closed 2025-02-28 by alfredgu001324