BeamNG/BeamNGpy #296: Add ProceduralObject with annotations

omeisosis, 2025-03-16:
Hello,
i was wondering if it would be possible to have a ProceduralObject  show up in annotation images from the Camera. As of right now, they appear with the BACKGROUND class, and i could not find a way to change this. 
I tried to chieve this in the world editor but it did not appear to have any effect, it would be most usefull if the annotaion-class could be assigned from within python.

Any hint into the right direction would be appreciated, thanks.

omeisosis, 2025-03-21:
It would be even better if I could use a custom annotation class with its own colour.

aivora-beamng, 2025-03-21:
Hi, I'm afraid there is no way how to do this currently, it is in our backlog. We will update this issue once this is resolved.

omeisosis, 2025-03-21:
Thank you for the Answer and continued support!

Quant14, 2025-12-09:
Hi @omeisosis, this issue will be resolved in the next release.

closed 2025-12-09 by Quant14