BeamNG/BeamNGpy #299: [Linux] errors

comejv, 2025-04-24:
Hello, 
I am on Ubuntu 24.04 trying to make beamng and beamngpy work (I am giving up) but I'll still report errors I've seen : 

BeamNG 0.35.5 with 1.32
With beamng already started with `-tcom` arg I would expect this to work but simulator doesn't start and python fails:
```
>>> b = BeamNGpy(host='localhost', port=25252, home='/home/come/Documents/BeamNG/BeamNG.tech.v0.35.5.0')
>>> b.open(launch=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "...python3.12/site-packages/beamngpy/beamng/beamng.py", line 197, in open
    self._load_system_info()
  File "...python3.12/site-packages/beamngpy/beamng/beamng.py", line 229, in _load_system_info
    info = self.system.get_info()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "...python3.12/site-packages/beamngpy/api/beamng/system.py", line 29, in get_info
    return self._send(data).recv()
           ^^^^^^^^^^^^^^^^
  File "...python3.12/site-packages/beamngpy/beamng/beamng.py", line 367, in _send
    return self.connection.send(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...python3.12/site-packages/beamngpy/connection/connection.py", line 199, in send
    raise BNGError("Cannot send, not connected to the simulator.")
beamngpy.logging.BNGError: Cannot send, not connected to the simulator.
```

With the `launch=True` parameter we now get this error even though the simulator itself launches properly:
```
>>> b = BeamNGpy(host='localhost', port=25252, home='/home/come/Documents/BeamNG/BeamNG.tech.v0.35.5.0')
>>> b.open(launch=True, args='-headless')
Error connecting to BeamNG.tech. 25 tries left.
[Errno 111] Connection refused
Traceback (most recent call last):
  File "...python3.12/site-packages/beamngpy/connection/connection.py", line 142, in connect_to_beamng
    self.skt = PrefixedLengthSocket(self.host, self.port)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...python3.12/site-packages/beamngpy/connection/prefixed_length_socket.py", line 58, in __init__
    self.skt.connect((host, port))
ConnectionRefusedError: [Errno 111] Connection refused
Error connecting to BeamNG.tech. 24 tries left.
```

BeamNG 0.34.3 with 1.31
with launch=False : same error as 0.35

with launch=True : works (but then beamng itself has lua errors)

Also quick question, why is setting up home not optionnal in linux but is in Windows? In linux if I don't give it the beamng home path it won't even try to open(). Relative paths won't work too.

comejv, 2025-04-25:
Thank you for your answer. I cannot test linux and 0.35 now but reading the lua sources it seems like the server is not opened with -tcom unless the legacy mode is on? Or is it started later?

```lua
M.onInit = function()
  setExtensionUnloadMode(M, 'manual')

  local cmdArgs = Engine.getStartingArgs()
  local legacyInit = false -- new versions of beamngpy use custom command to open the server, but we need to keep compatibility
  for i, v in ipairs(cmdArgs) do
    if v == '-rport' then
      legacyInit = true -- new version of beamngpy do not send this command-line option, the port is set in the openServer function
      tcomParams.port = tonumber(cmdArgs[i + 1]) or tcomParams.port
    elseif v == '-tcom-debug' then
      tcomParams.debug = true
    elseif v == '-no-tcom-debug' then
      tcomParams.debug = false
    elseif v == '-tcom-listen-ip' then
      tcomParams.ip = cmdArgs[i + 1]
    end
  end

  setup()
  if legacyInit then
    M.openServer()
  end
  refreshScenarioList()
end
```
And on windows + 0.34.2 + 1.31 I still get the cannot send error unless I explicitely start the server with `.\BeamNG.tech.x64.exe -lua "extensions.load('tech/techCore');tech_techCore.openServer(25252)" -console` in which case I can connect without issues. Which makes me think that with only -tcom the server is not started?

comejv, 2025-04-26:
I also found that when the host is not localhost or 127.0.0.1 BeamNGpy hangs until connexion is established until BeamNG starts, which is not the case with localhost.

In other words with distant host I can start beamngpy first, with localhost I cannot.

comejv, 2025-04-26:
I just realised I've been copying some lua files from version 0.33.3 to new version, so the errors come from me. I'll test again properly and report on monday.

comejv, 2025-04-30:
Yes the issue was coming from our side, sorry for bothering you!

The only issues remaining :
When using local host and starting beamng.py the open() function doesn't block, immediately tries to send data :
```
>>> from beamngpy import BeamNGpy
>>> b = BeamNGpy(host='127.0.0.1', port=25252)
>>> b.open(launch=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/beamng/beamng.py", line 197, in open
    self._load_system_info()
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/beamng/beamng.py", line 229, in _load_system_info
    info = self.system.get_info()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/api/beamng/system.py", line 29, in get_info
    return self._send(data).recv()
           ^^^^^^^^^^^^^^^^
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/beamng/beamng.py", line 367, in _send
    return self.connection.send(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/connection/connection.py", line 199, in send
    raise BNGError("Cannot send, not connected to the simulator.")
beamngpy.logging.BNGError: Cannot send, not connected to the simulator.
```

Trying to launch beamng.tech from beamngpy raises a "permission denied" error:
```
>>> b = BeamNGpy(host='127.0.0.1', port=25252, home="/home/come/Documents/BeamNG/BeamNG.tech.v0.35.5.0/")
>>> b.open(launch=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/beamng/beamng.py", line 194, in open
    self._start_beamng(extensions, *arg_list, **opts)
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/beamng/beamng.py", line 449, in _start_beamng
    binary = filesystem.determine_binary(home)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/xyzihsdhqpxsc4rc8wyk3wdjkx15s0lc-python3-3.12.8-env/lib/python3.12/site-packages/beamngpy/beamng/filesystem.py", line 73, in determine_binary
    if binary.exists():
       ^^^^^^^^^^^^^^^
  File "/nix/store/0l539chjmcq5kdd43j6dgdjky4sjl7hl-python3-3.12.8/lib/python3.12/pathlib.py", line 860, in exists
    self.stat(follow_symlinks=follow_symlinks)
  File "/nix/store/0l539chjmcq5kdd43j6dgdjky4sjl7hl-python3-3.12.8/lib/python3.12/pathlib.py", line 840, in stat
    return os.stat(self, follow_symlinks=follow_symlinks)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/home/come/Documents/BeamNG/BeamNG.tech.v0.35.5.0/BinLinux/BeamNG.tech.x64'
```

But I can work with that, I just need to start beamnpy first myself.

aivora-beamng, 2025-05-01:
Hi, the "permission denied" error could mean that either:
- `/home/come/Documents/BeamNG/BeamNG.tech.v0.35.5.0/BinLinux/BeamNG.tech.x64` does not exist
- `/home/come/Documents/BeamNG/BeamNG.tech.v0.35.5.0/BinLinux/BeamNG.tech.x64` exists, but the executable bit may be missing

closed 2025-05-05 by comejv