BeamNG/BeamNGpy #300: Can't connect to simulator

nbdy, 2025-05-19:
BeamNG.drive Alpha v0.35.5 (Steam)
BeamNGpy 1.32
Debian 12 (6.1.X, AMD64)
Python 3.12 via pyenv

I'm starting BeamNG with [what i found in the documentation](https://documentation.beamng.com/beamng_tech/arguments_and_settings/#connect-to-beamngpy)

```bash
./BeamNG.drive.x64 -tcom
```
(netstat tells me there is port 25252 in use when BeamNG is running, which already does not seem desired)

and am just trying to connect to beamng

```python
BEAMNG_PATH = "omitted"
bng = BeamNGpy('localhost', 25252, BEAMNG_PATH)
bng.open(launch=False)
```
(setting launch to True launches BeamNG, which tells me the path is correct, but that should not matter if there already is a process running)

which causes

```
beamngpy.logging.BNGError: Cannot send, not connected to the simulator.
```

Is there something I am doing wrong here?

aivora-beamng, 2025-05-21:
Hi,
BeamNGpy officially supports and is tested on BeamNG.tech only, so our support for BeamNG.drive is limited.

When you run with the `-tcom` argument, BeamNG starts the TCP server (by default on port 25252), so the netstat output suggests that part of the connection is working. Do you get any logs inside BeamNG about connection or connection errors?

If you are connecting to an existing instance of BeamNG, you don't need to fill the BEAMNG_PATH.

For more debug information, you can try `telnet 127.0.0.1 25252`, Wireshark/tshark or other network debugging tools. BeamNGpy itself has options for detailed logging (`log_communication` argument here: https://documentation.beamng.com/api/beamngpy/master/beamngpy.html#beamngpy.logging.config_logging). You can also try to use a different port to listen on with the `-tport` argument (https://documentation.beamng.com/beamng_tech/arguments_and_settings/#listening-on-other-network-interfaces).

nbdy, 2025-05-21:
Thanks for the quick response.
My goal is to use the Terrain_Importer as well as the OpenStreetMapImporter (which might have to get extended, that's why I forked the repo) to create maps from SRTM/DEM and OSM. Is this achievable with BeamNG.drive and beamngpy or can I stop here?

I've just noticed that I've written "netstat tells me there is port 25252" which is not right. The sentence should have included a "no". I've edited the Issue accordingly.
```bash
netstat -tulpn | grep 25252
```
does not output anything.

<details>
<summary>Here is the whole BeamNG.drive startup log when starting with `./BeamNG.drive.x64 -tcom`.</summary>

```text
nbdy@c0:/media/nbdy/safe/SteamLibrary/steamapps/common/BeamNG.drive/BinLinux$ ./BeamNG.drive.x64 -tcom
Shutdown handler: initialize.
  >>>>  [CURRENT_CONFIGURATION=SHIPPING=BNG_SHIPPING]  <<<<  
  0.90337|D|crashrpt| Installing crash reporter on path: /home/nbdy/.local/share/BeamNG.drive/0.35/temp/crashReports
  4.19913|I|engine| Virtual Filesystem: user path: /home/nbdy/.local/share/BeamNG.drive/0.35/
  4.19915|D|engine|  - user path reason: Default path
  4.19916|D|engine|  - user mods path: /home/nbdy/.local/share/BeamNG.drive/0.35/mods/
  4.19916|D|engine|  - game mods path: /media/nbdy/safe/SteamLibrary/steamapps/common/BeamNG.drive/content/
  4.19918|D|main| H2 3A0F69D03D098C1F2D09AB2CC58AD64385AB
>   4.24120|I|init| Initializing platform...
  4.30424|D|engine| initializing 113 modules (only displaying anything above 0.1 s)...
  4.30429|D|gfx| GFX Init:
  4.30430|D|gfx|    Null device found
  4.41054|D|fmod| FMOD API version: 2.02.17
  4.49040|D|sfx| Using default audio device: Yamaha AG06MK2 Analog Stereo
  4.55591|D|sfx| Created FMOD device 'Yamaha AG06MK2 Analog Stereo'
  4.55592|D|sfx|   CAPS_VoiceManagement
  4.55617|D|sfx|    Provider: FMOD
  4.55618|D|sfx|    Hardware: false
  4.55618|D|sfx|    Buffers: 16
[S_API] SteamAPI_Init(): SteamAPI_IsSteamRunning() did not locate a running instance of Steam.
[S_API] SteamAPI_Init(): Loaded '/home/nbdy/.local/share/Steam/linux64/steamclient.so' OK.
  4.56837|D|initialize| Could not initialize Steam functions
  4.56837|E|Steam| Could not initialize - Error code 1
  4.85906|I|GELua.| ============== GELUA VM loading ===============
  6.42057|D|GELua.gameplay_missions_progress.| Loaded 0 condition types from 0 files.
  6.43336|D|GELua.gameplay_missions_missions.| Loaded 2 procedural Mission Generators from 2 files.
  6.45865|D|GELua.gameplay_missions_unlocks.| Loaded 10 condition types from 4 files.
  6.51461|D|engine|  - GameEngineLua in 1.95 seconds
  6.51464|D|engine| ... all modules loaded in 2.210 seconds
  6.51489|D|init| OpenXR: Steam VR mode: disabled
  6.92219|D|gfx| Available adapters: 5
  6.92221|D|gfx|  - name:GPU (Vulkan), outputName:\\.\DISPLAY1, gfx:Vulkan, gpu:GPU
  6.92221|D|gfx|  - name:GPU (Vulkan), outputName:\\.\DISPLAY2, gfx:Vulkan, gpu:GPU
  6.92222|D|gfx|  - name:GPU (Vulkan), outputName:\\.\DISPLAY3, gfx:Vulkan, gpu:GPU
  6.92222|D|gfx|  - name:GPU (Vulkan), outputName:\\.\DISPLAY4, gfx:Vulkan, gpu:GPU
  6.92222|D|gfx|  - name:GFX Null Device, outputName:, gfx:, gpu:
  6.92223|D|gfx| Attempting to create GFX device 0: GPU (Vulkan) [with output \\.\DISPLAY1]
  6.92223|D|gfx| shader model: 5.0 , available video modes: 28
  6.92295|D|gfx| Device created, setting adapter and enumerating modes
  7.11607|D|vulkan| 20 supported Vk instance extensions: VK_EXT_display_surface_counter, VK_EXT_debug_utils, VK_EXT_debug_report, VK_EXT_acquire_xlib_display, VK_EXT_acquire_drm_display, VK_KHR_device_group_creation, VK_KHR_get_physical_device_properties2, VK_KHR_display, VK_EXT_direct_mode_display, VK_KHR_external_semaphore_capabilities, VK_KHR_get_display_properties2, VK_KHR_external_memory_capabilities, VK_KHR_portability_enumeration, VK_KHR_get_surface_capabilities2, VK_KHR_xcb_surface, VK_KHR_external_fence_capabilities, VK_KHR_surface, VK_KHR_surface_protected_capabilities, VK_KHR_wayland_surface, VK_KHR_xlib_surface
  7.11669|D|vulkan| The requested Vulkan instance extension does not appear to be supported, skipping extension: VK_NV_external_memory_capabilities
  7.11884|D|vulkan| 8 Layers: VK_LAYER_VALVE_steam_fossilize_64, VK_LAYER_VALVE_steam_overlay_64, VK_LAYER_VALVE_steam_fossilize_32, VK_LAYER_VALVE_steam_overlay_32, VK_LAYER_NV_optimus, VK_LAYER_MESA_device_select, VK_LAYER_INTEL_nullhw, VK_LAYER_MESA_overlay
  7.11886|D|vulkan| Attempting to create vulkan instance with the following 8 instance extensions: VK_KHR_get_physical_device_properties2, VK_EXT_debug_utils, VK_KHR_surface, VK_KHR_external_memory_capabilities, VK_KHR_external_fence_capabilities, VK_KHR_external_semaphore_capabilities, VK_KHR_surface, VK_KHR_xlib_surface
  7.15048|D|vulkan| Available vulkan devices:
  7.15050|D|vulkan|  - name:NVIDIA GeForce RTX 4060 Ti vendor:4318 version:2247999552 type:Discrete
  7.15050|D|vulkan|  - name:llvmpipe (LLVM 15.0.6, 256 bits) vendor:65541 version:1 type:Cpu
  7.15051|D|vulkan| Trying to locate a vulkan device with type: Discrete
  7.15052|D|vulkan|  - Found vulkan device with suitable type: name:NVIDIA GeForce RTX 4060 Ti vendor:4318 version:2247999552 type:Discrete
  7.16193|D|vulkan| 187 supported Vk device extensions: VK_AMD_buffer_marker, VK_NVX_image_view_handle, VK_NV_viewport_swizzle, VK_NV_viewport_array2, VK_NV_shader_subgroup_partitioned, VK_NV_shader_image_footprint, VK_NV_ray_tracing_motion_blur, VK_NV_ray_tracing_invocation_reorder, VK_NV_ray_tracing, VK_NV_optical_flow, V [... cut ...]
 22.41329|W|GELua.core_hardwareinfo.core_hwinfo| No desktop entry found. To get a BeamNG entry and (optionally) support for the beamng:// URL scheme, please run the script BinLinux/BeamNG_install_desktop_file.sh.
```

</details>

As suggested I have enabled communication logging via
```python
beamngpy.logging.config_logging([], False, logging.DEBUG, log_communication=True)
```
but there is no additional output.

I've also tried variations such as
```bash
./BeamNG.drive.x64 -tcom -tport 22222 -tcom-listen-ip {127.0.0.1, localhost, 10.0.0.X (Interface IP)}
```
and made sure the port that should be used is not already in use.

nbdy, 2025-05-21:
This does not seem to be an OS specific issue as I am facing the same problem on Windows (11) with BeamNG.drive 0.35.5 as well as the latest version of this lib.

aivora-beamng, 2025-05-22:
Thank you for the debugging and the log (that helped), I tested locally and I can confirm the `-tcom` argument doesn't work on BeamNG.drive (that is a bug, not intended). We will fix it next update.

Run BeamNG like this and the server should start:
```bash
./BeamNG.drive.x64 -lua "extensions.tech_techCore.openServer(25252)" 
```

You should be able to use the importers, AFAIK the code is included with the base BeamNG installation.

nbdy, 2025-05-22:
Thanks, that works.

Looking forward to upcoming updates, thanks for your and your teams work.