BeamNG/BeamNGpy #301: AI Controller

alfredgu001324, 2025-08-20:
Hi there,

Thanks for the great work and all the detailed documentation! I have one question related to the AI controller you have, what is the mechanism behind it? Is it some MPC or PID controller in essence and I am wondering how it works to follow waypoints/paths?

Thank you again and looking forward to your reply.

AbdelrahmanElsaidElsawy, 2025-09-10:
Hello @alfredgu001324,

Thank you for reaching out. Lateral control (steering wheel) is a pure pursuit controler. It follows a trajectory that is generated in real time, using geometry data from a map navigation graph (a navigation graph is like google maps for our maps). 

Longitudinal control (throttle and brake) is basically pid, following a target velocity. The target velocity is calculated from the velocity profile of the trajectory. The velocity profile is a backwards integration solver that takes into account the trajectory geometry (ex. cuvature) the road/vehicle traction limits, obstacles on the road (ex. other vehicles, traffic) and user specified parameters (desired acceleration envelope).

This is a good reference for the velocity profile solver: "Minimum Curvature Trajectory Planning and Control for an Autonomous Racecar" (2020) Heilmeier et al

There are some additional controllers for oversteer and understeer correction.

Best regards,
Abdelrahman

alfredgu001324, 2025-09-21:
Thank you so much for your reply! Really appreciate your help. I will look into this and let you know if I have further questions. Thanks!

closed 2025-09-21 by alfredgu001324