BeamNG/BeamNGpy #302: Configuring the BeamNG.AI to pass through multiple checkpoints

alessiogambi, 2025-09-08:
Currently, the beamng.ai can be configured to reach a specific checkpoint on the map. @aivora-beamng: Would it be possible to extend the interface to accept a list of checkpoints instead, with the semantics of reaching all of them one after the other? Currently, I need to force the vehicle to drive through a specific point before reaching the final checkpoint, and I realized that script.ai, with manually scripted trajectories, does not follow the lanes.

AbdelrahmanElsaidElsawy, 2025-09-09:
Hi @alessiogambi,

You can achieve this by using the following commands:

```python
<vehicle>.ai.set_waypoint("<waypoint_name>")
<vehicle>.ai.drive_in_lane(True)
```

Replace <vehicle> with your vehicle instance (e.g., ego_vehicle) and <waypoint_name> with the name of the waypoint you'd like the vehicle to reach (e.g., "junction1_wp28").

You can chain multiple set_waypoint calls in sequence if you want the vehicle to visit several waypoints. Setting drive_in_lane(True) ensures the vehicle follows the road lanes instead of cutting corners or taking direct paths, unlike script.ai.

Let me know if you need help looping over a list of waypoints or creating a route!

Best,  
Abdelrahman

alessiogambi, 2025-09-09:
Hi @AbdelrahmanElsaidElsawy , 
Thanks for the prompt reply. 

When you say chain the calls to set_waypoint what do you mean precisely?
If I call the method `set_waypoint` twice with two parameters, I have the impression that the second call overwrites the first one.

If I need to wait until the vehicle reaches the first waypoint before calling `set_waypoint` a second time, will the vehicle stop at the first waypoint and then restart when set to the second one? As this is *not* the behavior I need, which is that the vehicle keeps driving.

aivora-beamng, 2025-09-09:
Hi @alessiogambi, the previous answer is wrong, sorry for that. Please have a look at `/lua/vehicle/ai.lua`, there is the `driveUsingPath` function which should satisfy your needs. We will have a look at exposing it to Python, but you can already use it via the `vehicle.queue_lua_command` strategy.

```lua
ai.driveUsingPath{ wpTargetList = {'wp1', 'wp10'}, driveInLane = 'on', avoidCars = 'on', routeSpeed = 35, routeSpeedMode = 'limit', wpSpeeds = {wp1 = 10, wp2 = 40}, aggression = 0.3}
```

The documentation for the arguments is included in the source Lua file.

alessiogambi, 2025-09-10:
Hi @aivora-beamng , thanks for the clarification. 
I'll try your solution using `queue_lua_command` and report my findings here.
If it is worth, I'll open a feature request to expose that call in the Python API.

aivora-beamng, 2025-09-10:
We'll be shipping the `vehicle.ai.drive_using_waypoints` function in BeamNG.tech 0.37, thank you for the interest :)